dji-sdk/Onboard-SDK-ROS #413: Controlling Mavic Air 2

Houman-HM, 2020-11-05:
Dear DJI community,

I have a question regarding controlling the Mavic Air 2 drone using Onboard-SDK-ROS. Is it possible to control the actual drone using this package? If yes, how can one put multiple Mavic Air 2s on the same network and control them for swarm robotics applications.

dji-dev, 2020-11-06:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/40803'>Zendesk ticket #40803</a>:<br><br>I am sorry. Onboard SDK doesn't support Mavic air 2. In fact, consumer series drones do not support OSDK. The drone supports onboard SDK please refer to https://developer.dji.com/products/#!/onboard.

Houman-HM, 2020-11-07:
Thanks for the quick reply.
So, is there any other way to get connected to Mavic Air 2 WIFI network and send velocity commands to it, without using its own remote controller?

dji-dev, 2020-11-09:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/40803'>Zendesk ticket #40803</a>:<br><br>I am sorry. There is currently no way to meet your needs.

closed 2021-06-25 by dji-kyle