dji-sdk/Onboard-SDK-ROS #416: M300 waypoint2 error!

tuyaliang, 2020-11-13:
M300 waypoint2 error:

OSDK:4.0.1
ROS:4.0.1
Firmware = 3.4.8.56
baud_rate: 230400

`aiost@aiost:~/catkin_ws$ roslaunch dji_osdk_ros dji_vehicle_node.launch
... logging to /home/aiost/.ros/log/2f4866c8-2592-11eb-bb83-00044ba56dc2/roslaunch-aiost-16828.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.63:46059/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.9
 * /vehicle_node/acm_name: /dev/ttyACM0
 * /vehicle_node/align_time: False
 * /vehicle_node/app_id: 1093670
 * /vehicle_node/app_version: 1
 * /vehicle_node/baud_rate: 230400
 * /vehicle_node/enc_key: 7a4c81ac71d2e669e...
 * /vehicle_node/serial_name: /dev/ttyUSB0
 * /vehicle_node/use_broadcast: False

NODES
  /
    vehicle_node (dji_osdk_ros/dji_vehicle_node)

auto-starting new master
process[master]: started with pid [16838]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 2f4866c8-2592-11eb-bb83-00044ba56dc2
process[rosout-1]: started with pid [16849]
started core service [/rosout]
process[vehicle_node-2]: started with pid [16856]
EnableAd: 1
[3236737.973]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH9300BGBU3
[3236737.973]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.56
[3236737.973]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[3236737.973]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[3236738.973]STATUS/1 @ sendHeartbeatToFCTask, L1576: OSDK send heart beat to fc task created.
[3236739.174]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[3236739.178]STATUS/1 @ RegisterOSDInfoCallback, L471: register result of geting　FC ground station status pushing : 1

[3236739.178]STATUS/1 @ Firewall, L46: Firewall is initializing ...
[3236739.195]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH9300BGBU3
[3236739.195]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.56
[3236739.213]STATUS/1 @ AdvancedSensing, L110: Start bulk channel for M300's liveview!
[3236739.213]STATUS/1 @ AdvancedSensing, L117: Create task for M300's liveview!
[3236739.216]STATUS/1 @ AdvancedSensing, L125: Start bulk channel for M300's perception
[3236739.216]STATUS/1 @ AdvancedSensing, L132: Create task for M300's advanced sensing!
[3236739.216]STATUS/1 @ AdvancedSensing, L135: Advanced Sensing init for the M300 drone
[3236739.216]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.
[3236740.016]STATUS/1 @ init, L254: Start advanced sensing initalization
[3236740.016]STATUS/1 @ activate, L1309: osdk policy file updating(1) ......
[3236741.018]STATUS/1 @ activate, L1317: osdk policy file updating(2) ......
[3236741.023]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:0
[3236741.023]STATUS/1 @ RequestUploadPolicyFileHandle, L163: Upload policy file info md5 checksum and version
[3236741.025]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:1
[3236741.025]STATUS/1 @ RequestUploadPolicyFileHandle, L183: request upload policy file data: 0 0 200
[3236741.026]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:1
[3236741.026]STATUS/1 @ RequestUploadPolicyFileHandle, L183: request upload policy file data: 1 200 169
[3236741.027]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:2
[3236741.027]STATUS/1 @ RequestUploadPolicyFileHandle, L211: request upload policy file success
[3236742.018]STATUS/1 @ activate, L1329: version 0x3040838

[3236742.032]STATUS/1 @ activate, L1367: Activation successful

[ INFO] [1605259544.459512341]: VehicleNode Start
[ INFO] [1605259544.466394619]: Topic startup!
[ INFO] [1605259544.556468302]: Services startup!
[ INFO] [1605259544.604604285]: Use data subscription to get telemetry data!
[ INFO] [1605259544.604773212]: align_time_with_FC set to false. We will use ros time to time stamp messages!
[3236743.211]STATUS/1 @ verify, L244: Verify subscription successful.
[3236743.231]STATUS/1 @ startPackage, L345: Start package 3 result: 0.
[3236743.231]STATUS/1 @ startPackage, L347: Package 3 info: freq=100, nTopics=3.

[3236743.253]STATUS/1 @ startPackage, L345: Start package 2 result: 0.
[3236743.253]STATUS/1 @ startPackage, L347: Package 2 info: freq=50, nTopics=14.

[3236743.281]STATUS/1 @ startPackage, L345: Start package 1 result: 0.
[3236743.281]STATUS/1 @ startPackage, L347: Package 1 info: freq=5, nTopics=1.

[3236743.303]STATUS/1 @ removePackage, L457: Remove package 1 successful.
[3236743.341]STATUS/1 @ startPackage, L345: Start package 1 result: 0.
[3236743.342]STATUS/1 @ startPackage, L347: Package 1 info: freq=5, nTopics=12.

[3236743.361]STATUS/1 @ startPackage, L345: Start package 4 result: 0.
[3236743.361]STATUS/1 @ startPackage, L347: Package 4 info: freq=400, nTopics=1.

[ INFO] [1605259562.523653163]: called waypointV2SubscribeMissionStateCallback
[3236761.115]STATUS/1 @ RegisterMissionStateCallback, L452: register result of geting mission state pushing : 1

[ INFO] [1605259562.529289340]: called waypointV2SubscribeMissionEventCallback
[3236761.121]STATUS/1 @ RegisterMissionEventCallback, L433: register result of geting mission event pushing : 1

[ INFO] [1605259562.537416536]: called waypointV2GenerateActionsCallback
[ INFO] [1605259562.546121392]: called waypointV2InitSettingCallback
[3236761.143]STATUS/1 @ initWaypointV2, L1980: Init mission setting successfully!

[ INFO] [1605259563.568377160]: called waypointV2UploadMissionCallback
[3236762.167]ERRORLOG/1 @ uploadWaypointV2, L1995: Upload waypoint v2 mission ErrorCode:0x110000010038

[3236762.168]ERRORLOG/1 @ printErrorCodeMsg, L931: >>>>Error module   : WaypointV2
[3236762.168]ERRORLOG/1 @ printErrorCodeMsg, L932: >>>>Error message  : waypoint position exceed radius limit 
[3236762.168]ERRORLOG/1 @ printErrorCodeMsg, L933: >>>>Error solution : none
`

dji-dev, 2020-11-16:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/41026'>Zendesk ticket #41026</a>:<br><br>Hi. If you run SDK on ROS, the baud_rate should be larger than 921600.

tuyaliang, 2020-11-16:
Set baud_rate 921600 OK

closed 2020-11-16 by tuyaliang