dji-sdk/Onboard-SDK-ROS #417: M300 flight control take off failed!

tuyaliang, 2020-11-13:
M300 flight take off failed:

```
started roslaunch server http://192.168.1.63:33661/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.9
 * /vehicle_node/acm_name: /dev/ttyACM0
 * /vehicle_node/align_time: False
 * /vehicle_node/app_id: 1093670
 * /vehicle_node/app_version: 1
 * /vehicle_node/baud_rate: 230400
 * /vehicle_node/enc_key: 7a4c81ac71d2e669e...
 * /vehicle_node/serial_name: /dev/ttyUSB0
 * /vehicle_node/use_broadcast: False

NODES
  /
    vehicle_node (dji_osdk_ros/dji_vehicle_node)

auto-starting new master
process[master]: started with pid [17110]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f970e038-2592-11eb-a437-00044ba56dc2
process[rosout-1]: started with pid [17121]
started core service [/rosout]
process[vehicle_node-2]: started with pid [17128]
EnableAd: 1
[3237076.939]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH9300BGBU3
[3237076.939]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.56
[3237076.939]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[3237076.940]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[3237077.940]STATUS/1 @ sendHeartbeatToFCTask, L1576: OSDK send heart beat to fc task created.
[3237078.140]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[3237078.144]STATUS/1 @ RegisterOSDInfoCallback, L471: register result of geting　FC ground station status pushing : 1

[3237078.145]STATUS/1 @ Firewall, L46: Firewall is initializing ...
[3237078.159]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH9300BGBU3
[3237078.160]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.56
[3237078.183]STATUS/1 @ AdvancedSensing, L110: Start bulk channel for M300's liveview!
[3237078.183]STATUS/1 @ AdvancedSensing, L117: Create task for M300's liveview!
[3237078.184]STATUS/1 @ AdvancedSensing, L125: Start bulk channel for M300's perception
[3237078.184]STATUS/1 @ AdvancedSensing, L132: Create task for M300's advanced sensing!
[3237078.184]STATUS/1 @ AdvancedSensing, L135: Advanced Sensing init for the M300 drone
[3237078.184]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.
[3237078.984]STATUS/1 @ init, L254: Start advanced sensing initalization
[3237078.985]STATUS/1 @ activate, L1309: osdk policy file updating(1) ......
[3237079.986]STATUS/1 @ activate, L1317: osdk policy file updating(2) ......
[3237079.993]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:0
[3237079.993]STATUS/1 @ RequestUploadPolicyFileHandle, L163: Upload policy file info md5 checksum and version
[3237079.995]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:1
[3237079.995]STATUS/1 @ RequestUploadPolicyFileHandle, L183: request upload policy file data: 0 0 200
[3237079.997]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:1
[3237079.997]STATUS/1 @ RequestUploadPolicyFileHandle, L183: request upload policy file data: 1 200 169
[3237079.999]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:2
[3237079.999]STATUS/1 @ RequestUploadPolicyFileHandle, L211: request upload policy file success
[3237080.986]STATUS/1 @ activate, L1329: version 0x3040838

[3237080.997]STATUS/1 @ activate, L1367: Activation successful

[ INFO] [1605259883.427437214]: VehicleNode Start
[ INFO] [1605259883.443157666]: Topic startup!
[ INFO] [1605259883.547266901]: Services startup!
[ INFO] [1605259883.603002761]: Use data subscription to get telemetry data!
[ INFO] [1605259883.603193474]: align_time_with_FC set to false. We will use ros time to time stamp messages!
[3237082.218]STATUS/1 @ verify, L244: Verify subscription successful.
[3237082.238]STATUS/1 @ startPackage, L345: Start package 3 result: 0.
[3237082.238]STATUS/1 @ startPackage, L347: Package 3 info: freq=100, nTopics=3.

[3237082.260]STATUS/1 @ startPackage, L345: Start package 2 result: 0.
[3237082.260]STATUS/1 @ startPackage, L347: Package 2 info: freq=50, nTopics=14.

[3237082.291]STATUS/1 @ startPackage, L345: Start package 1 result: 0.
[3237082.291]STATUS/1 @ startPackage, L347: Package 1 info: freq=5, nTopics=1.

[3237082.308]STATUS/1 @ removePackage, L457: Remove package 1 successful.
[3237082.348]STATUS/1 @ startPackage, L345: Start package 1 result: 0.
[3237082.348]STATUS/1 @ startPackage, L347: Package 1 info: freq=5, nTopics=12.

[3237082.370]STATUS/1 @ startPackage, L345: Start package 4 result: 0.
[3237082.370]STATUS/1 @ startPackage, L347: Package 4 info: freq=400, nTopics=1.

[ INFO] [1605259897.648191015]: Obtain Control Authority Callback
[ INFO] [1605259897.721566726]: call takeoff service
Takeoff failed. Motors are not spinning.

```

dji-dev, 2020-11-16:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/41027'>Zendesk ticket #41027</a>:<br><br>Did you test in the simulator or in the real world? Did the plane actually take off when this error log appeared?

APACHE-THE-THIRD, 2020-11-23:
I'm observing the same issue when taking off and landing with m210v2. Plane actually takes off/lands, but log says it failed and subscriber calling the service is also getting an error. Timeout is too short in monitoredTakeoff and monitoredLanding methods calls?

APACHE-THE-THIRD, 2020-11-30:
Increasing baud rate to 1000000 fixed the problem in m210v2.