dji-sdk/Onboard-SDK-ROS #419: OSDK waypoint mission with RTK made by third party

wheelie96, 2020-11-16:
Hello there,

Now I'm using DJI OSDK-ROS 3.8 with Matrice 600 pro.
And I'm trying a waypoint mission on demo_mission.cpp in OSDK.

Here, a waypoint mission is executed based on the GPS information from the existing GPS antenna, but I would like to use a third-party RTK module and change it to execute the mission based on that RTK information.
The RTK module I want to use is u-blox's C94-M8P. It has the feature of being compatible with the ROS environment. The product homepage and ROS package are as follows.
https://www.u-blox.com/en/product/c94-m8p?lang=ja
https://github.com/KumarRobotics/ublox

sensor_msgs::NavSatFix gps_pos;
ros::Subscriber        gps_pos_subscriber;
void
gpsPosCallback(const sensor_msgs::NavSatFix::ConstPtr& msg)
{
  gps_pos = *msg;
}
Regarding demo_mission.cpp, the above part is related to GPS information, so I think it is necessary to change this part, but what should I do when using the above ROS package for RTK?

Thank you;

dji-dev, 2020-11-17:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/41102'>Zendesk ticket #41102</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Because you are using a third-party RTK module, OSDK（-ROS） cannot support obtaining third-party RTK information by subscribing to drone telemetry data. You can consider combining the specific modules you use and modifying the implementation of ROS to directly obtain your RTK data instead of drone telemetry data.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

wheelie96, 2020-11-18:
Thank you for your comment.
Now, I have another questions about the waypoint mission.

Waypoint missions generate all waypoints in advance and run the mission after they are uploaded. At this time, when moving from the current waypoint to the next waypoint during the mission, is there a possibility that the position uploaded in advance and the position of the destination will deviate due to the error of GPS during movement?

Also, GPS is usually used to acquire location information during mission execution, but is it possible to change location information acquisition to a third-party RTK instead? In this case, do I need to make any changes such as connecting the RTK module directly to the flight controller?

I apologize for the inconvenience, but please answer.

dji-dev, 2020-11-19:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/41102'>Zendesk ticket #41102</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Waypoint missions generate all waypoints in advance and run the mission after they are uploaded. At this time, when moving from the current waypoint to the next waypoint during the mission, is there a possibility that the position uploaded in advance and the position of the destination will deviate due to the error of GPS during movement?

-->The drone positioning is based on GPS signals. If there is a GPS error, there may be an error in the flight position of the aircraft.

Also, GPS is usually used to acquire location information during mission execution, but is it possible to change location information acquisition to a third-party RTK instead? In this case, do I need to make any changes such as connecting the RTK module directly to the flight controller?
     -->The M600 flight controller only supports the D-RTK module to obtain RTK signals. If use a third-party RTK module, developers can evaluate the use of SDK to connect to RTK module and develop related positioning functions by themselves. The signal cannot be directly received by the flight controller.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

wheelie96, 2020-11-19:
Thank you for your comment.

I understood that instead of connecting the RTK module directly to the flight controller, I need to connect it to the module via the onboard CPU and SDK.
My current system configuration is to connect the onboard CPU to the API port of the flight controller via a TTL cable and execute the mission from OSDK-ROS.
Therefore, we believe that we will connect the RTK module to the onboard CPU and send RTK information to the flight controller via the onboard CPU.
And, as I mentioned earlier, I would like to use RTK for the position judgment of the flight controller, but first I would like to know the position judgment mechanism when using GPS.
So I would like to ask, does OSDK-ROS have executable files (.cpp, .h, etc.) necessary for the flight controller to determine the position by GPS during mission execution?
I think I can change from GPS to RTK by changing the contents of that file, is that wrong?

Alternatively, if you fly with the default product specifications that do not connect the CPU to the flight controller, the flight controller will receive information directly from the antenna without going through a framework like ROS.
Therefore, does it mean that there is no special executable file required for GPS position determination?

I apologize for the inconvenience, but please answer.

lorenzmhe, 2020-11-19:
Hi @wheelie96

Just giving you a bit of context before you get an answer from DJI.

In the current design of the OSDK it is not possible to supply GPS information through the OSDK or ROS-OSDK. Whenever you start a mission it is executed by the flight controller and it will use the built in GPS.

dji-dev, 2020-11-20:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/41102'>Zendesk ticket #41102</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Because the third-party RTK module cannot be directly connected to the flight controller, the OSDK（ROS） cannot be used to correct the positioning information(or change the GPS to RTK) of the flight controller during the execution of the mission.

If you use OSDK to develop an integrated RTK module by yourself, the positioning is realized by the OSDK application, and then call the OSDK flight control API (for example: Control::flightCtrl). In this case, it is not feasible to directly use the waypoint function provided by OSDK（ROS）, because it will use the built-in GPS positioning of the drone.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

wheelie96, 2020-12-26:
Excuse me for your comment after a long time.

About "it is not feasible to directly use the waypoint function", is there a possibility that it can be used indirectly if it cannot be used directly?

Please answer.

tyrant63, 2022-10-16:
ROS is Not applicable