dji-sdk/Onboard-SDK-ROS #420: How are the waypoint's altitude and longitude determined?

wheelie96, 2020-11-24:
Hello there,

Now I have a simple question about the program "demo_mission.cpp" .
In this program, at the subroutine "generateWaypointsPolygon" , the waypoint's latitude and longitude are determined as follows.

 // Iterative algorithm
  for (int i = 1; i < num_wp; i++)
  {
    WayPointSettings  wp;
    WayPointSettings* prevWp = &wp_list[i - 1];
    setWaypointDefaults(&wp);
    wp.index     = i;
    wp.latitude  = (prevWp->latitude + (increment * cos(i * extAngle)));
    wp.longitude = (prevWp->longitude + (increment * sin(i * extAngle)));
    wp.altitude  = (prevWp->altitude + 1);
    wp_list.push_back(wp);
  }

Here, why are wp.latitude and wp.longitude calculated this way?
What theory is this calculation based on?
I can't understand why the value "increment * cos or sin (i * ext Angle)" is added to the latitude and longitude of the previous waypoint. What is the value of "increment"?

Sorry to trouble you, but please let me know.

dji-dev, 2020-11-25:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/41410'>Zendesk ticket #41410</a>:<br><br>Hi. This is the method for generating polygonal waypoints. The num_wp indicates how many sides in the Polygon. The increment is the side length.
For example, you can set num_wp to 6 and draw on paper to see if it is a regular hexagon.

wheelie96, 2020-11-25:
Thank you for your comment.

But I have already known that the program draws the polygon and can change the number of sides and the length of sides.
So I want to know why this waypoint program can draw the polygon.

Is it some mathematical theory-based calculation that determines the latitude and longitude of the next waypoint by adding increment multiplied by the external angle cos or sin to the latitude and longitude of the previous waypoint? Or is it a calculation method that is not based on any theory, which was forcibly determined to draw polygons?
Also, regarding the calculation "increment = 0.000001 / C_PI * 180", I can see that the unit is converted from radians to degrees, but I do not understand why it is given by such an expression.

Sorry to trouble you, but please let me know.

dji-dev, 2020-11-26:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/41410'>Zendesk ticket #41410</a>:<br><br>Hi. I don't know the theoretical basis for this. I think you can easily prove it by drawing a picture. Because in waypoint mission for ROS, latitude and longitude are expressed in angles. So the increment needs to be converted from angle to radians.

wheelie96, 2020-11-26:
Thank you for your comment.

Leaving my question aside, Are you not the creator of this ”demo_mission.cpp" program? 
If you are not, I want a comment from the creator.