dji-sdk/Onboard-SDK-ROS #423: [M300] The video quality from main camera stream is bad

dangviethieu, 2020-11-29:
Hi there. I'm getting this issue when stream from main camera:`rosrun dji_osdk_ros camera_stream_node`
The video quality is bad, often vague and flicker.
Any one can help me? Thank you 🙇 
[image]
[image]

dji-dev, 2020-11-30:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/41540'>Zendesk ticket #41540</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

About this issue， the OSDK ros has updated an SDL rendering display scheme,you can update this version and install SDL (sudo apt-get install libsdl2-dev) and run the camera-stream-node， option：b to verify it.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dangviethieu, 2020-11-30:
@dji-dev I have been use newest source code from OSDK master branch, and installed SDL, but the issue still occurs 
[image]

dangviethieu, 2020-11-30:
The error is only occurs with main camera.

dji-dev, 2020-12-01:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/41540'>Zendesk ticket #41540</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

I have reproduced the problem and submitted it to the relevant engineer for further analysis and confirmation. If there is any result, we will promptly feedback and communicate with you.
Sorry for the inconvenience caused to your related development work

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-dev, 2020-12-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/41540'>Zendesk ticket #41540</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

About the issue :The video quality from main camera stream is bad
The current confirmation information is related to the decoding performance of the OSDK running platform.
What platform do you run OSDK on? Is it convenient to provide hardware platform, CPU architecture and performance related information?

We tested it on X86(i7) and DJI manifold 2G to verify that it can be used normally.
If possible, it is recommended that you use DJI manifold 2G to verify whether the abnormality still occurs.

And about the error promts, this error does not affect normal use.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dangviethieu, 2020-12-09:
@dji-dev I also tested it on DJI manifold 2G and CPU (i7), the issue still occurs.
Is this is problems of version ffmpeg/libav? I used the last version.
Could you share me screen when you test successfully with main camera. Or I need to reinstall DJI manifold 2G?

dji-dev, 2020-12-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/41540'>Zendesk ticket #41540</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

So if you are currently using manifold 2G, pixelated video images will also appear?

The error will still appear, this is related to the codec video data and does not affect the decoding display.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dangviethieu, 2020-12-09:
@dji-dev So how can I resolve this issue, because I really need your solution?
As I said before, the issue is only occurs with main camera.

dji-dev, 2020-12-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/41540'>Zendesk ticket #41540</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry, it may need to be clarified. The current question you are asking is divided into two parts.
1. The terminal log prompts error, this does not affect the use at present, you can ignore this error message first.
2. Vague and flicker appear when decoding the main camera. It is currently confirmed that the problem is related to the performance of the platform used. We use the latest master branch version of OSDK-ROS on manifold 2G. The camera_stream_node : b option has optimized this problem (with SDL).
so whether you use manifold-2G and using the latest OSDK-ROS will still appear vague and flicker phenomenon?

In addition：
3, about this issue, It is only occurs with main camera(H20). Because H20 uses GDR encoding format, different from FPV (IDR). And we have adopted SDL in the ROS version to optimize it, currently it is recommended to use manifold 2G and the latest ROS version to use SDL display.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

LuossD, 2020-12-31:
I want to know whether the transmission of uav data and camera images after OSDK development can be transmitted to Windows must through MSDK？