dji-sdk/Onboard-SDK-ROS #424: H264 different stream between platforms: M210 vs M300

lorenzave, 2020-12-01:
Hi support,
we are using directly the h264 stream to ease ROS communication and use HW accelerated transcoding. We can use the M300 streams (H20T and XT2) with no problem. When we try doing the same using the m210v2rtk and an XT2 the output in the topic is very different. 
With the M300 we get a steady 30Hz and a properly populated message. With the M210 we get something between 38 and 45Hz and some of the messages are "empty" with a data size of 6 bytes. We have been trying to filter these out because they break our pipeline but sometimes we get several nearly empty messages and it's too much for it to recover. Is this intended? The same camera should output different streams depending on the platform? 

I attach two [rosbags](https://drive.google.com/file/d/1r2Etxbjgz5SZWcZrgZ85Bby7EFxSLCSW/view?usp=sharing) with the different outputs and the same camera. Everything is with the last firmware. There is some topic renaming for the M300 but it is remapped from the original DJI topic.

Thanks

dji-dev, 2020-12-02:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/41624'>Zendesk ticket #41624</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The same camera should output the same on different platforms, but it is need to confirm whether the video stream obtained by the OSDK is normal. Have you tried to save the H264 file to compare the difference?
If so, it is recommended that you can provide the H264 raw stream files of the corresponding platform respectively, and we will further confirm.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

lorenzave, 2020-12-02:
Hi Support,
I attach the raw h264 files from the liveview sample [here](https://drive.google.com/file/d/1F700zOpNwNCppH1OoiNbNk8XwI9W5l3u/view?usp=sharing).
It seems that both files are fine and decodeable. The issue seems related to the rostopic publishing as we get different behavior between platforms. Try and do some rosbag play on the rosbags I sent previously and see how different the messages are there.

Thanks for the feedback and have a nice day.

dji-dev, 2020-12-04:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/41624'>Zendesk ticket #41624</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry for the link to the rosbags and H264 stream files you provided, it is not accessible here. If the H264 file is normal, may need to further investigate the ROS application. If it is convenient, we recommend that you send the problem and corresponding information to dev@dji.com, and describe the operation steps in detail. We will try to test this issue and verify it.
including the drone firmware version and camera firmware version.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

lorenzave, 2020-12-04:
Hi Support,
Ill try uploading them here to github. You have both rosbags and both h264 raw files recorded using the liveview sample.
Please inspect the difference between the ROS messages between plaforms. H264 seems to be fine, from my perspective it seems there is a difference in behavior in how the H264 callback works in each platform. M300 streams at 30Hz (rostopic wise) and M210 (38-45 Hz).
[m300+XT2_rosbag.zip](https://github.com/dji-sdk/Onboard-SDK-ROS/files/5641499/m300%2BXT2_rosbag.zip)
[m210v2rtk+XT2_rosbag.zip](https://github.com/dji-sdk/Onboard-SDK-ROS/files/5641505/m210v2rtk%2BXT2_rosbag.zip)
[RawH264files.zip](https://github.com/dji-sdk/Onboard-SDK-ROS/files/5641507/RawH264files.zip)

I hope you can access these within github.

Best regards.

dji-dev, 2020-12-07:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/41624'>Zendesk ticket #41624</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

I can already access the files you submitted to github.It seems to be related to the implementation of OSDK and M210 models and supported camera models.
We have submitted this information to the relevant engineers for further confirmation and analysis. If there is any progress or communication needs, we will Communicate with you again.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

lorenzave, 2020-12-07:
Hi Support,
Thanks. As said before we can easily work with the M300 h264 topic and the XT2 camera. Once we try it out in the M210s it breaks because of the data "chopping" is we pipe those topics to a raw file and read both then it's fine. I guess it's how the callbacks in the OSDK subscribers work for the different platforms.

I look forward to hearing from you.

Best regards.

closed 2021-02-25 by lorenzave

dji-dev, 2021-07-07:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/41624'>Zendesk ticket #41624</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Regarding this issue, we have done a backtracking again, and the comparison test results on M300 and M210 are synchronized with you as follows:
1.M300

    OSDK-ROS4.1.0: FPV: 30HZ; Z30: 30HZ; XT2: 30HZ

    OSDK4.1.0: FPV:30HZ; Z30:30HZ; XT2:30HZ

2.M210:

      OSDK-ROS4.1.0: FPV: 45HZ; Z30: 15HZ; XT2: 13HZ

      OSDK4.1.0: FPV:45HZ; Z30:15HZ; XT2:12-13HZ

Preliminary determination is related to drone hardware，unfortunately, because the M210 is no longer updated and maintained. I noticed that you have closed this issue, the above information is for reference only.If it is needed, suggest to use a more stable drone M300 to develop this function.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>