dji-sdk/Onboard-SDK-ROS #426: Waypoint mission behavior

gggoh1, 2020-12-04:
I am using v4.0.1 of OSDK with M210 V2 RTK. I ran the sample mission_node waypoint mission, but it moves back a little bit when it reaches each waypoint.
Have anyone any idea about the cause of this?
[image]

dji-dev, 2020-12-04:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/41721'>Zendesk ticket #41721</a>:<br><br>Hi. This is because there will be inertia when the drone reaches the waypoint.

gggoh1, 2020-12-05:
@dji-dev thank you, I change idle_velocity

dji-dev, 2020-12-07:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/41721'>Zendesk ticket #41721</a>:<br><br>Yes. Setting a smaller idle_velocity can reduce overshoot.

closed 2020-12-08 by gggoh1