dji-sdk/Onboard-SDK-ROS #428: can not get RTK informations

potBagMeat, 2020-12-14:
dear developer:

  I recently developed M300 with RTK, I open rtk on the RC, but it shows it was disconnected  when I connect my Manifold 2-C to M300. I am sure I purchased the service, I `rostopic echo /dji_osdk_ros/rtk_connection_status`, it always shows 1. is there anything else can I do if I want to enable RTK, should I add [setRTKEnableSync](https://developer.dji.com/onboard-api-reference/functions_s.html) function in code?

My drone : M300
OSDK version: OSDK-ROS 4.0.1

dji-dev, 2020-12-15:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/42065'>Zendesk ticket #42065</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The TOPIC：rtk_connection_status， The TOPIC: rtk_connection_status, it shows whether the drone supports RTK, you are using M300 RTK, so it always shows 1.

you can enable RTK with RC or call API: setRTKEnableSync for M300, and it is recommended to confirm the RTK set on RC and select the RTK you use.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

potBagMeat, 2020-12-15:
Thanks a lot, sir.

closed 2020-12-15 by potBagMeat