dji-sdk/Onboard-SDK-ROS #429: Building packages without installing Onboard-SDK systemwide

i-sheremet, 2020-12-15:
Hello,

I am not able to find a way how to compile **Onboard-SDK-ROS** without installing **Onboard-SDK** systemwide (inside `/usr/local/`).

I compile **Onboard-SDK** like this:

`cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/home/my_user/Onboard-SDK/install`
`make install`

But can't able to include this path while building **Onboard-SDK-ROS**. I tried
`export CMAKE_PREFIX_PATH=$CMAKE_PREFIX_PATH:/home/my_user/Onboard-SDK/install/include`
`export PATH=$PATH:/home/my_user/Onboard-SDK/install/include`
`catkin_make`
but it still fails to find **dji_osdk** headers.

Any advice on that? It's really frustrating that the only way to build **Onboard-SDK-ROS** is by merging DJI headers and binaries with the Linux ones.

dji-dev, 2020-12-16:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/42124'>Zendesk ticket #42124</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Because DJI OSDK ROS is an application based on OSDK Linux, OSDK Linux lib and header file need to be installed before compile the ROS.
and you have modified the library path installed by OSDK, you need to configure the corresponding path in ROS compile file.
such as:
add the directories before compile the ROS:
include_directories(
 ...
 /your/folder/include
)

link_directories(
 ...
 /your/folder/lib
 )

In addition, you can refer to the default installation to /usr/lib

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

guilhermelawless, 2021-01-20:
A rough example, add to CMakeLists.txt and during compilation pass the cmake variable `-DDJI_SDK_INSTALL_DIR=/your/djisdk/install/dir`

```cmake
if(DEFINED DJI_SDK_INSTALL_DIR)
  include_directories(${DJI_SDK_INSTALL_DIR}/include)
  link_directories(${DJI_SDK_INSTALL_DIR}/lib)
  message(STATUS "Local installation: DJI SDK at ${DJI_SDK_INSTALL_DIR}")
else()
  message(AUTHOR_WARNING "DJI_SDK_INSTALL_DIR not set, will look in system directories")
endif()
```

closed 2021-01-20 by i-sheremet