dji-sdk/Onboard-SDK-ROS #435: Improvement of altitude by installing other sensors in waypoint mission

wheelie96, 2020-12-24:
Hello there;

Now I use OSDK-ROS 3.8 and Matrice 600 pro, and I added a change to "demo_mission.cpp" like below.
In line 155 at "demo_mission.cpp",
wp.altitude  = (prevWp->altitude + 1) ⇒ wp.altitude  = (prevWp->altitude).

By this change, UAV's altitude is kept constant during the waypoint mission.
This time, UAV's altitude is determined by a barometric pressure sensor equipped in UAV, but I know the accuracy of its sensor is poor.
So I want to improve the altitude by installing some sensors (vision sensor, LiDAR, and so on) to UAV, is that possible?
Is it possible to send advanced correction information such as vision sensor or LiDAR to the flight controller during waypoint mission execution?

Thank you;

dji-dev, 2020-12-25:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/42466'>Zendesk ticket #42466</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The DJI  OSDK (ROS)  can obtain telemetry data througt subscribtion or broadcast function, and the different TOPIC data from the specific sensor, you can refet to the API reference to choose the TOPIC:
https://developer.dji.com/onboard-api-reference/group__telem.html

eg:
TOPIC_ALTITUDE_FUSIONED
 [image]​

2、So I want to improve the altitude by installing some sensors (vision sensor, LiDAR, and so on) to UAV, is that possible?
Is it possible to send advanced correction information such as vision sensor or LiDAR to the flight controller during waypoint mission execution?
   -->Sorry, the DJI drone currently does not support inputting external sensor data to the flight controller through the SDK.
You can evaluate external sensor data processing, and then control the drone through OSDK attitude control . It may not be possible to directly use OSDK control functions, such as waypoint missions.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/nR5HtXo1PuVZ4ZiarVVp7pDY6' target='zd_attach'>inline-693713943.png</a>

wheelie96, 2020-12-26:
Thank you for your quick comment.

So, about "OSDK attitude control", where is that program like .cpp or .h?
Is OSDK-ROS 3.8 include it?
And, about "It may not be possible to directly use OSDK control functions, such as waypoint missions.", is there a possibility that it can be used indirectly if it cannot be used directly?

Please answer.

dji-dev, 2020-12-28:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/42466'>Zendesk ticket #42466</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

DJI OSDK ROS is developed based on DJI OSDK (Linux). The flight control API needs to find the corresponding code of OSDK (Linux),
please refer to:
https://github.com/dji-sdk/Onboard-SDK/blob/3.9/osdk-core/api/inc/dji_control.hpp

eg:
 void flightCtrl(CtrlData data);

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team