dji-sdk/Onboard-SDK-ROS #436: dji_sdk roslaunch failed at outdoor environment

wheelie96, 2020-12-25:
Hello there;

Before, when I executed a command "roslaunch dji_sdk sdk.launch" at outdoor environment using Matrice 600 pro and OSDK-ROS 3.8 on onboard computer, I failed to launch.

I got some message like a below picture when I did roslaunch at outdoor;
[image]
"Enable API control", "Baudrate", "TX and RX pin", "some program occupied serial port", and "permission" are indicated as factor of launch error in this picture, but they were ok when I checked on that time.

However, curiously, it succeeds at indoor environment that "roslaunch dji_sdk sdk.launch", like a below picture.
[image]
I did not change "Enable API control", "Baudrate", and etc. conditions between indoors and outdoors.
I wondered if its difference is that there is internet or not, so I do roslaunch at indoor after disconnecting internet, but I succeeded it. 

Is it possible for such a phenomenon to occur?
I want to get good advise.

Thank you;

dji-dev, 2020-12-28:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/42498'>Zendesk ticket #42498</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

According to the error display, the serial port communication is abnormal. This should not necessarily be related to indoor or outdoor. Please check whether the serial cable is loose during the movement. In addition, due to normal communication, you can replace the serial port cable to test whether the serial communication is stable. It is recommended to use the FT232 serial port module.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

wheelie96, 2020-12-29:
Thank you for your comment.

Yesterday, I took the drone outdoors and tried the outdoor roslaunch again.
At this time, I first tried roslaunch as it was carried out, but it still failed.
So, next I checked the looseness of the serial cable as you say, but there is no looseness in the TX pin and RX pin connected to the A3 flight controller, and there is no looseness in the USB terminal connected to the onboard CPU.
Then, after disconnecting the TX pin, RX pin, and USB terminal, I reconnected and tried roslaunch, but it still failed.

Next, after exchanging the ttl serial cable indoors, I started the drone and confirmed that roslaunch succeeded.
However, once I shut down the drone and CPU, took it outdoors, restarted it, and roslaunch failed. At this time, the cable was not loose as before, and even if I unplugged the cable and reconnected it, roslaunch failed.
Then, I took it indoors with the drone and CPU turned on in a state where roslaunch failed outdoors, and roslaunch again, but it failed. However, once I restarted the drone and CPU, it succeeded in roslaunch.
Therefore, I succeeded in roslaunching after taking it outdoors with the power on without shutting down. However, when I checked the ROS topic of gps_position outdoors, it was supposed to be delivered at 50Hz, but the number of Hz gradually decreased to 30Hz and 20Hz. And when I killed ROS outdoors and then roslaunch again, it failed this time.
After that, as before, I took it indoors with the drone and CPU turned on, and when I roslaunch again, it failed, but once I restarted the drone and CPU, roslaunch succeeded. Then, I replaced the ttl serial cable with the one that was originally connected and roslaunch successfully.

It's a strange phenomenon, but that's all I tried yesterday.
Thank you for your advice.

dji-dev, 2020-12-29:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/42498'>Zendesk ticket #42498</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The result of your test is a bit strange. Have you tried to change the baud rate, such as lowering the baud rate to 230400, and directly use the Linux demo to verify?

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

wheelie96, 2020-12-29:
Thank you for your quick reply.

You said "directly use the Linux demo to verify", does that mean I start the program directly to get gps_position without starting ROS? And what is the Linux demo?
I want to see "dji_sdk/gps_position" topic when I launch dji_sdk, but I don't know how to see just that topic without launching ROS.

dji-dev, 2020-12-29:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/42498'>Zendesk ticket #42498</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

No, DJI OSDK supports linux platform and ROS , the ROS is an application developed based on OSDK Linux，directly use OSDK Linux to compile and run the demo.
https://github.com/dji-sdk/Onboard-SDK/tree/4.0.1

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

wheelie96, 2020-12-29:
Thank you for your quick reply.

At the beginning, I tried roslaunch after change the baud rate at the indoor environment earlier, but I cannot roslaunch.
I changed the baud rate on DJI Assistant 2 and sdk.launch, is it right?
[image]
[image]

And, you said that I should use OSDK Linux, not OSDK-ROS, but I want to record the information by using "rosbag record -a",
so I think I cannot do it if I use OSDK Linux.
In addition, now I use the program of waypoint mission in dji_sdk_demo at OSDK-ROS 3.8, but there is not such program in OSDK Linux.

I don't think it's a fundamental solution without roslaunch, so please help me.

dji-dev, 2020-12-30:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/42498'>Zendesk ticket #42498</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

1、I changed the baud rate on DJI Assistant 2 and sdk.launch, is it right?
     -->Yes, change the baud rate on DJI Assistant 2 and sdk.launch, and ensure that the two values are the same.

2、And, you said that I should use OSDK Linux, not OSDK-ROS, but I want to record the information by using "rosbag record -a",
so I think I cannot do it if I use OSDK Linux.
In addition, now I use the program of waypoint mission in dji_sdk_demo at OSDK-ROS 3.8, but there is not such program in OSDK Linux.

-->The above suggestion to use the Linux demo is only for testing and troubleshooting this abnormal issue, because the current abnormal information you provide is related to serial communication, but the reason cannot be confirmed temporarily. Using Linux can use a lower baud rate (230400), and the use of ROS needs to support a higher baud rate, using a low baud rate may cause communication abnormalities, which is not convenient for troubleshooting.

About M600, OSDK 3.9 can be choosed for testing.

     https://github.com/dji-sdk/Onboard-SDK/tree/3.9

    If convenient, it is recommended to do some serial communication checks.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

wheelie96, 2020-12-30:
Thank you for your comment.

Now, because it is snowy at outdoor, I cannot takeoff M600pro.
Can I do troubleshooting without M600pro takeoff?
And, because I have no time, so please tell me the detail of how to do test such as serial communication check using OSDK Linux 3.9.
What the program should I execute in OSDK Linux 3.9?

dji-dev, 2020-12-30:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/42498'>Zendesk ticket #42498</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Not need to test takeoff, you can use any demo to start the test serial communication, such as： djiosdk-telemetry-sample

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

wheelie96, 2021-01-02:
Hello there.

2 days ago, I tried the test serial communication (djiosdk-telemetry-sample).
The behavior of the serial test was unstable both indoors and outdoors.
I succeeded several times indoors, but often failed in serial communication. I think the communication was more unstable than when I used roslaunch. However, when serial communication failed indoors, the communication could be successful if the test was executed while holding down the TX pin and RX pin on the A3 flight controller side.
In the outdoors, communication often failed as in the case of roslaunch, but communication was successful only once.

I used two ttl cables for the test this time, but the behavior of the above indoor and outdoor serial tests did not change in either case.

From the results of this serial test, I personally don't understand the reason for the error.
Possible reasons are poor serial communication contact or indoor and outdoor temperature differences?
The ttl cable I'm currently using is a very cheap one purchased from amazon, but does the difference in performance depending on the price affect serial communication?
In addition, the temperature in the experimental environment is about 15 ° C indoors and -5 ° C or less outdoors, so the temperature difference is 20 ° C or more. Does this large temperature difference affect serial communication?

The screen of the serial communication result is shown below. One for success and one for failure.
[image]
[image]

So please help me.

dji-dev, 2021-01-04:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/42498'>Zendesk ticket #42498</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The result seems to be related to the stability of serial communication. Your screenshot shows that the baud rate of 921600 is used. You can try to lower the baud rate, such as 230400. As for the stability impact of serial communication, further verification and investigation may be required in your abnormal environment.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

wheelie96, 2021-01-04:
Thank you for your comment.

I also tried the serial test on the same day with a baud rate of 230400.
I changed the baud rate in DJI Assistant 2 and Userconfig.txt and ran it.
At that time, I tested it indoors several times, but I remember that all of them failed to communicate.