dji-sdk/Onboard-SDK-ROS #438: about gimbal control of OSDK-ROS4.0.1 interface

potBagMeat, 2020-12-29:
Hi, dear developers

recently I have been developing some functions about gimbal. I need to use speed control of gimbal. However, I found that in the gimbal_camera_control_node in OSDK-ROS4.0.1, it only used angle control rather than angle speed control. I found speed control of gimbal in the OSDK-ROS-obsoleted interface. I dont know whether there will be same function OSDK-ROS4.0.1.

Thanks in advance.

dji-dev, 2020-12-29:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/42576'>Zendesk ticket #42576</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry, the current gimbal manager function supported by OSDK (ROS) 4.x does not support speed control of gimbal.
If you have requirements for relevant functionality, It is recommended that you click on the links on the right side and fill in the relevant demand information,  https://formcrafts.com/a/dji-developer-feedback-en
In order to better match your requirements, try to include the following information when filling in：
1. Desired functions and requirements
2. Company/organization profile
3. Project descriptions that use this feature requirement.
4. Personal background or brief introduction.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

potBagMeat, 2020-12-29:
@dji-dev 
Thanks for  your reply.

closed 2020-12-29 by potBagMeat