dji-sdk/Onboard-SDK-ROS #448: AdvancedSensing Errors when launching vehicle node without Root

nZac, 2021-01-25:
Without Root:

```
$ ls -la /dev/ttyACM0 /dev/ttyUSB0
crw-rw---- 1 root dialout 166, 0 Jan 25 15:49 /dev/ttyACM0
crw-rw---- 1 root dialout 188, 0 Jan 25 15:49 /dev/ttyUSB0

$ roslaunch dji_osdk_ros dji_vehicle_node.launch
started core service [/rosout]
process[vehicle_node-2]: started with pid [8082]
EnableAd: 1
[998627.812]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH8L00B2JT3
[998627.812]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.44
[998627.812]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[998627.813]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[998628.813]STATUS/1 @ sendHeartbeatToFCTask, L1546: OSDK send heart beat to fc task created.
[998629.013]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[998629.014]STATUS/1 @ RegisterOSDInfoCallback, L471: register result of geting　FC ground station status pushing : 1

[998629.015]STATUS/1 @ Firewall, L46: Firewall is initializing ...
[998629.053]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH8L00B2JT3
[998629.053]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.44
[998629.066]ERRORLOG/1 @ AdvancedSensing, L108: Failed to initialize USB Bulk Linker channel for liveview!
[998629.066]ERRORLOG/1 @ AdvancedSensing, L115: Failed to create task for liveview!
[998629.068]ERRORLOG/1 @ AdvancedSensing, L123: Failed to initialize USB Bulk Linker channel for perception!
[998629.069]ERRORLOG/1 @ AdvancedSensing, L130: Failed to create task for advanced sensing!
[998629.069]STATUS/1 @ AdvancedSensing, L135: Advanced Sensing init for the M300 drone
[998629.069]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.
[998629.870]STATUS/1 @ init, L254: Start advanced sensing initalization
[998629.870]STATUS/1 @ activate, L1309: osdk policy file updating(1) ......
[998630.871]STATUS/1 @ activate, L1317: osdk policy file updating(2) ......
[998630.875]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:0
[998630.876]STATUS/1 @ RequestUploadPolicyFileHandle, L163: Upload policy file info md5 checksum and version
[998630.877]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:1
[998630.878]STATUS/1 @ RequestUploadPolicyFileHandle, L183: request upload policy file data: 0 0 200
[998630.881]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:1
[998630.881]STATUS/1 @ RequestUploadPolicyFileHandle, L183: request upload policy file data: 1 200 169
[998630.883]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:2
[998630.883]STATUS/1 @ RequestUploadPolicyFileHandle, L211: request upload policy file success
```

With root
```
started core service [/rosout]
process[vehicle_node-2]: started with pid [7954]
EnableAd: 1
[998592.332]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH8L00B2JT3
[998592.332]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.44
[998592.332]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[998592.333]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[998593.333]STATUS/1 @ sendHeartbeatToFCTask, L1546: OSDK send heart beat to fc task created.
[998593.533]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[998593.534]STATUS/1 @ RegisterOSDInfoCallback, L471: register result of geting　FC ground station status pushing : 1

[998593.535]STATUS/1 @ Firewall, L46: Firewall is initializing ...
[998593.572]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH8L00B2JT3
[998593.572]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.44
[998593.586]STATUS/1 @ AdvancedSensing, L110: Start bulk channel for M300's liveview!
[998593.586]STATUS/1 @ AdvancedSensing, L117: Create task for M300's liveview!
[998593.588]STATUS/1 @ AdvancedSensing, L125: Start bulk channel for M300's perception
[998593.588]STATUS/1 @ AdvancedSensing, L132: Create task for M300's advanced sensing!
[998593.589]STATUS/1 @ AdvancedSensing, L135: Advanced Sensing init for the M300 drone
[998593.589]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.
[998594.390]STATUS/1 @ init, L254: Start advanced sensing initalization
[998594.390]STATUS/1 @ activate, L1309: osdk policy file updating(1) ......
[998595.392]STATUS/1 @ activate, L1317: osdk policy file updating(2) ......
[998595.398]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:0
[998595.398]STATUS/1 @ RequestUploadPolicyFileHandle, L163: Upload policy file info md5 checksum and version
[998595.401]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:1
[998595.401]STATUS/1 @ RequestUploadPolicyFileHandle, L183: request upload policy file data: 0 0 200
[998595.404]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:1
[998595.404]STATUS/1 @ RequestUploadPolicyFileHandle, L183: request upload policy file data: 1 200 169
[998595.407]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:2
[998595.408]STATUS/1 @ RequestUploadPolicyFileHandle, L211: request upload policy file success
```

What permissions does the vehicle node require in order to use advanced sensing?

dji-dev, 2021-01-26:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/43371'>Zendesk ticket #43371</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Please add the DJI USB Device refering to the link:
https://developer.dji.com/document/e34c3fc4-dc0e-4cbf-b7c0-589b6e941a01

 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/odkSN93J8uHblWf2Kj9CAVx5K' target='zd_attach'>inline-507045324.png</a>

nZac, 2021-02-02:
This does fix the issue, thank you!

closed 2021-02-02 by nZac