dji-sdk/Onboard-SDK-ROS #449: About "dji_sdk/flight_control_setpoint_ENUvelocity_yawrate"

sakpancar, 2021-02-17:
Hi guys,
I need to know how to get the DJI Matrice 100 topic "dji_sdk / flight_control_setpoint_ENUvelocity_yawrate". For example: while setting this topic, how to check if the data on the x, y and z axes are correct. Does a hardware on the Matrice 100 control the amount of velocities in these x, y, and z axes? Like IMU for example. Or do you need a DJI Guidance for this control?

dji-dev, 2021-02-18:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/43980'>Zendesk ticket #43980</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The TOPIC： dji_sdk / flight_control_setpoint_ENUvelocity_yawrate， used to control the drone with VELOCITY via OSDK API:
void
DJISDKNode::flightControlVxVyVzYawrateCallback(
  const sensor_msgs::Joy::ConstPtr& pMsg)
{
  uint8_t flag = (Control::VERTICAL_VELOCITY |
                  Control::HORIZONTAL_VELOCITY |
                  Control::YAW_RATE |
                  Control::HORIZONTAL_GROUND |
                  Control::STABLE_ENABLE);
  float vx        = pMsg->axes[0];
  float vy        = pMsg->axes[1];
  float vz        = pMsg->axes[2];
  float yawRate  = pMsg->axes[3];

  flightControl(flag, vx, vy, vz, yawRate);
}

Regarding your question:
while setting this topic, how to check if the data on the x, y and z axes are correct?
    -->Sorry, I did not fully understand your question. The parameters x/y/z are the command parameters given to the flight controller by the developer to control the drone, that is, the drone will fly according to the incoming speed parameters. The "x, y，and z axes are correct" you described What does mean?

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

sakpancar, 2021-02-22:
Thank you very much for answering my question.

But I want some technical information about Matrice 100. I use DJI Matrice 100 indoor (without GPS). Normally, a device cannot understand how speed it is walking or rising in indoor environments without GPS.

My question is: After the speed information of any coordinate sent to matrice 100 using the topic "dji_sdk / flight_control_setpoint_ENUvelocity_yawrate", what equipment is the feedback received (IMU, guidance sensor etc.).

dji-dev, 2021-02-23:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/43980'>Zendesk ticket #43980</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

DJI UAV does not provide a complete flight plan in a non-GPS environment. When using OSDK in a non-GPS environment, developers need to use the attitude control API to control the UAV flight.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

sakpancar, 2021-02-23:
Thank you very much for your patience. There is one last question. So is this "dji_sdk / flight_control_setpoint_ENUvelocity_yawrate" required for DJI Guidance to use with Matrice 100? can i use this ROS Topic with Matrice 100 without DJI Guidance?

dji-dev, 2021-02-23:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/43980'>Zendesk ticket #43980</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Regarding Guidance, DJI no longer provides technical support，you can use the OSDK and OSDK ROS for M100.
Note:
     The M100 only supports OSDK 3.9 and below, OSDK 3.9 and OSDK ROS 3.8 is recommaned.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team