dji-sdk/Onboard-SDK-ROS #450: there are multiple ros services return nothing?

SuperBigHui, 2021-02-19:
Hi buddy
I am using the branch 4.1 and i found there are multiple ros services always return false so i check the source code.
You may check the function behind in dji_vehicle_node_mission_services.cpp:
waypointV2StartMissionCallback
waypointV2StopMissionCallback
waypointV2PauseMissionCallback
waypointV2ResumeMissionCallback
...

I can't judge if the call is success which may result in serious accidents.

dji-dev, 2021-02-19:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/44027'>Zendesk ticket #44027</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The ROS demo will return the call result
eg : waypointV2StartMissionCallback
 [image]

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/Uc9TZssPmjjaqX7dPEiEyEaVP' target='zd_attach'>inline683849580.png</a>

SuperBigHui, 2021-02-19:
so... there is no sentence followe that? like "return response.result" or sth?

dji-dev, 2021-02-19:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/44027'>Zendesk ticket #44027</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Okay, thank you for your feedback. We understand this issue and have submitted it for review. It will not affect normal use at this time. You can add the corresponding code first.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team