dji-sdk/Onboard-SDK-ROS #451: I can't execute the sample : waypointV2_node twice

SuperBigHui, 2021-02-19:
Hi again
A error occurred when i execute the waypointV2_node twice in DJISimulator.
the log are as follow:
[ INFO] [1613707761.415688859]: called waypointV2UploadActionCallback
[3095025.415]STATUS/1 @ ActionsEncode, L374: upload_action_ID:0
[3095025.415]STATUS/1 @ ActionsEncode, L374: upload_action_ID:1
[3095025.415]STATUS/1 @ ActionsEncode, L374: upload_action_ID:2
[3095025.415]STATUS/1 @ ActionsEncode, L374: upload_action_ID:3
[3095025.415]STATUS/1 @ ActionsEncode, L374: upload_action_ID:4
[3095025.415]STATUS/1 @ ActionsEncode, L374: upload_action_ID:0
[3095025.415]STATUS/1 @ ActionsEncode, L374: upload_action_ID:1
[3095025.415]STATUS/1 @ ActionsEncode, L374: upload_action_ID:2
[3095025.415]STATUS/1 @ ActionsEncode, L374: upload_action_ID:3
[3095025.415]STATUS/1 @ ActionsEncode, L376: total_len:108
[3095025.417]ERRORLOG/1 @ uploadWaypointV2Actions, L2166: Upload waypoint v2 actions ErrorCode:0x110000400001

[3095025.418]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : WaypointV2
[3095025.418]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : the ID of Action is duplicated. 
[3095025.418]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : none

So Must I relaunch the dji_vehicle_node before uploading a new wp mission?

dji-dev, 2021-02-19:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/44031'>Zendesk ticket #44031</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

This is because action ID is stored in the vehicle node and uploaded repeatedly. If you use the demo directly, you may need to modify the code, and you need to re-initialize the action variable when re-uploading.
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/qTQz3WtdZBE5XxOSQ3IEowGXM' target='zd_attach'>inline-1511241689.png</a>

SuperBigHui, 2021-02-19:
OK, thanks for your comment.