dji-sdk/Onboard-SDK-ROS #455: Unable to get camera stream M210 RTK+X7

Avistadrone, 2021-02-23:
Greetings,

I am unable to get the camera stream from Zenmuse X7 mounted in the Matrice 210 RTK. When I try to start the camera stream it responds with error but it doesn't give me any information.
The issue happens with the following setup:
1、M210 RTK v2(firmware: v01.00.710) ,
2、baudrate：921600
3、OSDK4.0.1 + OSDK-ROS 4.0.1
4、roslaunch dji_osdk_ros dji_vehicle_node.launch and rosservice call setup_camera_stream 1 1
started roslaunch server http://anuc:36005/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.16
 * /vehicle_node/acm_name: /dev/ttyACM0
 * /vehicle_node/align_time: False
 * /vehicle_node/app_id: 1073520
 * /vehicle_node/app_version: 1
 * /vehicle_node/baud_rate: 921600
 * /vehicle_node/enc_key: 1ce9987d181096cb7...
 * /vehicle_node/serial_name: /dev/ttyUSB0
 * /vehicle_node/use_broadcast: True

NODES
  /
    vehicle_node (dji_osdk_ros/dji_vehicle_node)

ROS_MASTER_URI=http://anuc:11311

process[vehicle_node-1]: started with pid [24828]
EnableAd: 1
[3476567.885]ERRORLOG/1 @ initVehicle, L162: Failed to initialize ACM Linker channel!
[3476567.896]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1DADFCQ001000T
[3476567.896]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.3.44
[3476567.896]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[3476567.896]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[3476568.896]STATUS/1 @ sendHeartbeatToFCTask, L1576: OSDK send heart beat to fc task created.
[3476569.096]STATUS/1 @ Control, L40: The control class is going to be deprecated.It will be better to use the FlightController class instead!
[3476569.096]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[3476569.097]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[3476569.097]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[3476569.097]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[3476569.097]STATUS/1 @ initDJIHms, L900: DJI HMS is not supported on this platform!

[3476569.097]STATUS/1 @ init, L247: USB is not plugged or initialized successfully. Advacned-Sensing will not run.
[3476569.097]STATUS/1 @ activate, L1329: version 0x304032C

[3476569.115]STATUS/1 @ activate, L1367: Activation successful

[ INFO] [1614089306.135922211]: VehicleNode Start
[ INFO] [1614089306.136257360]: Topic startup!
[ INFO] [1614089306.206148397]: Services startup!
[ INFO] [1614089306.228577651]: Use legacy data broadcast to get telemetry data!
[3476570.252]STATUS/1 @ verify, L244: Verify subscription successful.
[3476570.275]STATUS/1 @ startPackage, L345: Start package 1 result: 0.
[3476570.275]STATUS/1 @ startPackage, L347: Package 1 info: freq=50, nTopics=6.

anuc@anuc:~$ rosservice call setup_camera_stream 1 1
ERROR: service [/setup_camera_stream] responded with an error:

lorenzave, 2021-02-23:
It seems you are using the wrong format to call a service. The proper way is the following.
`rosservice call /setup_camera_stream "cameraType: 0
start: 0" 
`
It is important you keep the spaces and format. You should be able to autocomplete these fields if you sourced the ros ws properly.

dji-dev, 2021-02-24:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/44169'>Zendesk ticket #44169</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The log shows the ACM device not been recognized, please comfirm the dual-A USB cable has been connected and works fine.
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/eFjvWtJ3k5TNfTMJWi1Zn6B9g' target='zd_attach'>inline-1425013760.png</a>

Avistadrone, 2021-02-24:
@lorenzave In Kinetic can be called in both ways, actually the problem was the ACM0, no mather how a call the service that I can't get the stream, thank you.

@dji-dev As you pointed out, the ACM0 connection was the problem, now I can connect and Advance sensing is activated, however everytime I call the service /setup_camera_stream I get connection time out as a result.
I created the udev rules for the ACM0 and tried with and without chmod the port ACM0. The cable that I am using it's the same that works perfectly with DJI assistant 2.

The X7 firmware is the following: 01.07.00.45.

anuc@anuc:~$ roslaunch dji_osdk_ros dji_vehicle_node.launch 
started roslaunch server http://anuc:42811/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.16
 * /vehicle_node/acm_name: /dev/ttyACM0
 * /vehicle_node/align_time: False
 * /vehicle_node/app_id: 1073520
 * /vehicle_node/app_version: 1
 * /vehicle_node/baud_rate: 921600
 * /vehicle_node/enc_key: 1ce9987d181096cb7...
 * /vehicle_node/serial_name: /dev/ttyUSB0
 * /vehicle_node/use_broadcast: True

NODES
  /
    vehicle_node (dji_osdk_ros/dji_vehicle_node)

auto-starting new master
process[master]: started with pid [5595]
ROS_MASTER_URI=http://anuc:11311

setting /run_id to 04029238-7676-11eb-9aaf-1c697a64ba43

process[rosout-1]: started with pid [5608]
started core service [/rosout]
process[vehicle_node-2]: started with pid [5622]
EnableAd: 1
[3540746.029]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1DADFCQ001000T
[3540746.029]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.3.44
[3540746.029]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[3540746.029]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[3540747.029]STATUS/1 @ sendHeartbeatToFCTask, L1576: OSDK send heart beat to fc task created.
[3540747.229]STATUS/1 @ Control, L40: The control class is going to be deprecated.It will be better to use the FlightController class instead!
[3540747.229]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[3540748.495]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[3540748.495]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[3540748.495]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[3540748.495]STATUS/1 @ initDJIHms, L900: DJI HMS is not supported on this platform!

[3540748.509]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1DADFCQ001000T
[3540748.509]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.3.44
[3540748.509]STATUS/1 @ AdvancedSensing, L145: Advanced Sensing init for the M210 drone
[3540748.509]STATUS/1 @ init, L49: Looking for USB device...

[3540748.521]STATUS/1 @ init, L65: Found 7 USB devices, identifying DJI device...

[3540748.521]STATUS/1 @ init, L83: Found a DJI device...

[3540748.521]STATUS/1 @ init, L96: Attempting to open DJI USB device...

[3540748.521]STATUS/1 @ init, L134: ...DJI USB device started successfully.

[3540748.523]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.
[3540749.024]STATUS/1 @ init, L254: [3540749.024]STATUS/1 @ adv_pthread, L46: adv pthread created !!!!!!!!!!!!!!!!!!!!!!!
[3540749.024]STATUS/1 @ adv_pthread, L48: adv pthread running !!!!!!!!!!!!!!!!!!!!!!!
Start advanced sensing initalization
[3540749.024]STATUS/1 @ activate, L1329: version 0x304032C

[3540749.047]STATUS/1 @ activate, L1367: Activation successful

[ INFO] [1614153486.068504713]: VehicleNode Start
[ INFO] [1614153486.068795242]: Topic startup!
[ INFO] [1614153486.232684465]: Services startup!
[ INFO] [1614153486.271906623]: Use legacy data broadcast to get telemetry data!
[3540750.308]STATUS/1 @ verify, L244: Verify subscription successful.
[3540750.329]STATUS/1 @ startPackage, L345: Start package 1 result: 0.
[3540750.329]STATUS/1 @ startPackage, L347: Package 1 info: freq=50, nTopics=6.

ERRORLOG/1Connect to MAIN_CAMERA failed, Error: Connection setup failure: connection time out.

ERRORLOG/1Initialize MAIN_CAMERA failed
Double check USB connection or re-plug in USB cable.

ERRORLOG/1Connect to FPV_CAMERA failed, Error: Connection setup failure: connection time out.

ERRORLOG/1Initialize FPV_CAMERA failed
Double check USB connection or re-plug in USB cable.

ERRORLOG/1Connect to MAIN_CAMERA failed, Error: Connection setup failure: connection time out.

ERRORLOG/1Initialize MAIN_CAMERA failed
Double check USB connection or re-plug in USB cable.

ERRORLOG/1Connect to FPV_CAMERA failed, Error: Connection setup failure: connection time out.

ERRORLOG/1Initialize FPV_CAMERA failed
Double check USB connection or re-plug in USB cable.

dji-dev, 2021-02-24:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/44169'>Zendesk ticket #44169</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

​Please check the network with "ping 192.168.42.2" on the Onboard Computer.
If the network is unavailable, please check the ACM device IP with "ifconfig"
and then set the IP :
ifconfig [eth] down
ifconfig [eth] 192.168.42.3 up

NOTE:
1、[eth] --> ACM device network name
2、For M210，the DJI Assistant2 and OSDK cannot be connected to the drone at the same time. When using the OSDK USB communication, it is need to ensure that the Onboard computer is pinging 192.168.42.2 and the network is reachable.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

Avistadrone, 2021-02-24:
I rebooted both drone and onboardPC and now I am able to ping and thus get the stream.

I am closing this issue.

Thank you very much!

closed 2021-02-24 by Avistadrone