dji-sdk/Onboard-SDK-ROS #456: Unable to view stereo camera images M210 RTK v2 ( OSDK-ROS 4.1& OSDK 4.1)

AbhinandanVellanki, 2021-03-03:
Hello,

I am unable to view the stereo images being published over the topic **'/dji_osdk_ros/stereo_240p_front_left_images/'** or **'/dji_osdk_ros/stereo_240p_front_right_images/'** when running the ROS version of the OSDK. I am able to successfully run and view the rectified stereo images by running `rosrun dji_osdk_ros stereo_vision_depth_perception_node m210_stereo_param.yaml 
` and then choosing the option to display rectified stereo images. 

However, I need to calibrate my stereo cameras and update the m210_stereo_param.yaml file to get good depth perception results. I am trying to use the ros camera_calibration package (http://wiki.ros.org/camera_calibration/Tutorials/StereoCalibration) and this needs the stereo cameras' images to be published on two separate topics. I have attached some screenshots of me running the codes:

RUNNING OSDK-ROS 4.1:
[image]
[image]

VISUALIZING ROS TOPICS:
[image]

CALLING REQUIRED TOPIC:
[image]

MORE TOPICS:
[image]

CANNOT VISUALIZE EVEN IN RVIZ:
[image]

Can someone please help out by telling me how to get the camera stream messages to publish on these topics so that i can calibrate my stereo cameras using the Camera Calibration package. Additionally, any alternate methods to calibrate the stereo cameras would also be greatly appreciated :)

Thanks in advance!!

ShreyasManjunath, 2021-03-09:
Hi,

Did you call the camera service?
rosservice call '/stereo_vga_subscription' '{vga_freq: 0, front_vga: 1, unsubscribe_vga: 0}'

regards,
Shreyas

AbhinandanVellanki, 2021-03-10:
> Hi,
> 
> Did you call the camera service?
> rosservice call '/stereo_vga_subscription' '{vga_freq: 0, front_vga: 1, unsubscribe_vga: 0}'
> 
> If you are willing to subscribe to 240p front stereo topic for ros stereo calibration, it wont work. I am experiencing the same.
> 
> regards,
> Shreyas

Hi Shreyas,

Yes that worked!

Thank you