dji-sdk/Onboard-SDK-ROS #457: fatal error: dji_vehicle.hpp: no such file or directory

user651, 2021-03-18:
I cloned the current master version of the repository and followed the instructions to build the osdk. When executing `sudo make -j 4 install` i get the following error message: `fatal error: dji_vehicle.hpp: no such file or directory`. Is there any hint on what may be wrong?
I also appended the full log 
[ros_make.log](https://github.com/dji-sdk/Onboard-SDK-ROS/files/6165890/ros_make.log)

thank you in advance

dji-dev, 2021-03-19:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/45192'>Zendesk ticket #45192</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

OSDK ROS based on the OSDK linux ,so you should install the OSDK lib.
Please refer to the website:
https://developer.dji.com/document/e34c3fc4-dc0e-4cbf-b7c0-589b6e941a01
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/kNpRvk9xZiQ7Dv2lv1dvUpIpg' target='zd_attach'>inline233156685.png</a>

fatbringer, 2022-01-07:
Hi must the OSDK folder be placed inside the src folder inside catkin_ws ??

abdouakhad, 2022-08-16:
Yes It can be placed their

ricardohalfeld, 2024-02-27:
I'm having the same issue. This you solve this by placing the `Onboard-SDK` next to `Onboard-SDK-ROS` like this:

```
catkin_ws
├── build
├── devel
├── logs
└── src
    ├── Onboard-SDK
    └── Onboard-SDK-ROS
```

Even so it doesn't find `dji_vehicle.hpp`. Any further help will be appreciated.

ricardohalfeld, 2024-03-01:
I realized using the `raspberry-pi` support branch is what makes the dji_vehicle.* go away. Changing to the latest release branch brings them back, but causes issues with `ffmpeg`.