dji-sdk/Onboard-SDK-ROS #462: Enabling h264 stream on M210 hangs indefinitely in 4.1.0

lorenzmhe, 2021-04-19:
We have successfully been using 4.0.1 since shortly after its release. Recently, we upgraded to 4.1.0 and have started seeing the service call to enable h264 stream randomly block indefinitely. The frequency is 1-2 failures per 10 flights. In comparison, we haven't seen any issues with M300 in over 100 flights.

To recover we have had to kill the node process and do `rosnode cleanup` or even reboot the drone. 

**Setup**
- M210V2 01.00.0710
- XT2 01.00.00.34

** Failing service call**
`rosservice call /dji_osdk_ros/setup_camera_h264 "request_view: 0
start: 1"`

** Log file **
In this log file the stream was enabled and disabled a number of times. Note that is says the camera is already streaming at the button where the blocking happens. This is not the case; the camera is not streaming, and the service call blocks indefinitely.
[vehicle_node-6-stdout.log](https://github.com/dji-sdk/Onboard-SDK-ROS/files/6337008/vehicle_node-6-stdout.log)

dji-dev, 2021-04-20:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/46673'>Zendesk ticket #46673</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Could you please provide the XT2 Firmware(the camera firmvare: v01.00.00.34) ?
please comfirm and update to the lastest firmware to verify this issue:
https://dl.djicdn.com/downloads/Zenmuse%20XT%202/20200508/Zenmuse_XT%202_Release_Notes_en_.pdf
 [image]​

and please comfirm both OSDK(linux) and OSDK ROS has been upgraded to 4.1.0.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/mC6DPMLJOQiJ3delhQtUWfQDS' target='zd_attach'>inline1102599945.png</a>

lorenzmhe, 2021-04-20:
Thank you for your rapid response.

As said in the original post:

Camera 1 Firmware version: 01.00.00.34
OSDK(linux): 4.1.0
OSDK ROS: 4.1.0

dji-dev, 2021-04-21:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/46673'>Zendesk ticket #46673</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

I have test this issue with OSDK master branch and ROS master branch(orignal demo), stream was enabled and disabled, but failed to reproduce.
result:
 [image]

version:
 [image]

The log you provided, ​it seems that the thread stoped failed and not create pthread to receive stream, could you please provide more info about how to reprodce this issue, the better using the orignal demo to operate.
 [image]
 [image]​
​
Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
​<br><a href='https://djisdksupport.zendesk.com/attachments/token/Ndg7TcrRTDIEDAmLih1mgVMRe' target='zd_attach'>inline1592548630.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/pIE6WOfAKCHyeHDaLqj3r1Cr0' target='zd_attach'>inline-210594680.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/qcQGp381B4xaE7uw6L2iS2RFM' target='zd_attach'>inline1057252569.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/mFYt87CXqf4HIdH0SzsLrEbaT' target='zd_attach'>inline-1133830896.png</a>

lorenzmhe, 2021-04-21:
Unfortunately, as initially stated the failures are random/periodic.

We are using the provided vehicle_node and have exposed the async funtions to take off and land. 

The flow is:
1. Start camera h264 stream
2. Takeoff
3. Control with software joystick for 2 minutes
4. Land
5. Stop camera h264 stream
6. Repeat from step 1 until the error occurs

We have never seen the failure on the first start of the stream after boot of the drone. Also, it only happens when the drone flies in between start and stop of the stream.

dji-dev, 2021-04-21:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/46673'>Zendesk ticket #46673</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

From the information currently provided, it is recommended to troubleshoot the receiving stream thread exception during repeated start and stop of the camera stream.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
​