dji-sdk/Onboard-SDK-ROS #463: m300 waypointV2upload失败

guojing-gj, 2021-04-21:
SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.17
 * /vehicle_node/acm_name: /dev/ttyACM0
 * /vehicle_node/align_time: False
 * /vehicle_node/app_id: 1072401
 * /vehicle_node/app_version: 1
 * /vehicle_node/baud_rate: 1000000
 * /vehicle_node/enc_key: b1f9132a9638f17b5...
 * /vehicle_node/serial_name: /dev/ttyUSB0
 * /vehicle_node/use_broadcast: False

NODES
  /
    vehicle_node (dji_osdk_ros/dji_vehicle_node)

ROS_MASTER_URI=http://localhost:11311

process[vehicle_node-1]: started with pid [19106]
EnableAd: 1
[4014760.388]STATUS/1 @ parseDroneVersionInfo, L1126: Device Serial No. = 1ZNDH7800AB14R
[4014760.388]STATUS/1 @ parseDroneVersionInfo, L1128: Firmware = 3.4.8.69
[4014760.388]STATUS/1 @ functionalSetUp, L280: Shake hand with drone successfully by getting drone version.
[4014760.389]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[4014761.389]STATUS/1 @ sendHeartbeatToFCTask, L1580: OSDK send heart beat to fc task created.
[4014761.589]STATUS/1 @ Control, L40: The control class is going to be deprecated.It will be better to use the FlightController class instead!
[4014761.589]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[4014761.606]STATUS/1 @ Firewall, L45: Firewall is initializing ...
[4014761.606]STATUS/1 @ Firewall, L65: osdk policy file updating(1) ......
[4014762.607]STATUS/1 @ Firewall, L73: osdk policy file updating(2) ......
[4014762.613]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:0
[4014762.614]STATUS/1 @ RequestUploadPolicyFileHandle, L234: Upload policy file info md5 checksum and version
[4014762.616]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[4014762.616]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 0 0 200
[4014762.619]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[4014762.619]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 1 200 169
[4014762.621]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:2
[4014762.621]STATUS/1 @ RequestUploadPolicyFileHandle, L282: request upload policy file success
[4014763.608]STATUS/1 @ firewallTask, L112: firewall task created ...
[4014763.632]STATUS/1 @ parseDroneVersionInfo, L1126: Device Serial No. = 1ZNDH7800AB14R
[4014763.632]STATUS/1 @ parseDroneVersionInfo, L1128: Firmware = 3.4.8.69
[4014763.647]STATUS/1 @ AdvancedSensing, L110: Start bulk channel for M300's liveview!
[4014763.647]STATUS/1 @ AdvancedSensing, L117: Create task for M300's liveview!
[4014763.649]STATUS/1 @ AdvancedSensing, L125: Start bulk channel for M300's perception
[4014763.649]STATUS/1 @ AdvancedSensing, L132: Create task for M300's advanced sensing!
[4014763.649]STATUS/1 @ AdvancedSensing, L135: Advanced Sensing init for the M300 drone
[4014763.649]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.
[4014764.449]STATUS/1 @ init, L255: Start advanced sensing initalization
[4014764.449]STATUS/1 @ activate, L1333: version 0x3040845

[4014764.468]STATUS/1 @ activate, L1371: Activation successful

[ INFO] [1618922468.784799999]: VehicleNode Start
[ INFO] [1618922468.785047830]: Topic startup!
[ INFO] [1618922468.890926596]: Services startup!
[ INFO] [1618922468.910517904]: Use data subscription to get telemetry data!
[ INFO] [1618922468.910549096]: align_time_with_FC set to false. We will use ros time to time stamp messages!
[4014765.631]STATUS/1 @ verify, L244: Verify subscription successful.
[4014765.645]STATUS/1 @ startPackage, L345: Start package 3 result: 0.
[4014765.645]STATUS/1 @ startPackage, L347: Package 3 info: freq=100, nTopics=3.

[4014765.668]STATUS/1 @ startPackage, L345: Start package 2 result: 0.
[4014765.668]STATUS/1 @ startPackage, L347: Package 2 info: freq=50, nTopics=15.

[4014765.688]STATUS/1 @ startPackage, L345: Start package 1 result: 0.
[4014765.688]STATUS/1 @ startPackage, L347: Package 1 info: freq=5, nTopics=1.

[4014765.711]STATUS/1 @ removePackage, L457: Remove package 1 successful.
[4014765.749]STATUS/1 @ startPackage, L345: Start package 1 result: 0.
[4014765.749]STATUS/1 @ startPackage, L347: Package 1 info: freq=5, nTopics=12.

[4014765.768]STATUS/1 @ startPackage, L345: Start package 4 result: 0.
[4014765.768]STATUS/1 @ startPackage, L347: Package 4 info: freq=400, nTopics=1.

[ INFO] [1618922474.870888944]: called waypointV2GenerateActionsCallback
[ INFO] [1618922474.874820489]: called waypointV2InitSettingCallback
[4014771.587]STATUS/1 @ initWaypointV2, L2196: Init mission setting successfully!

[ INFO] [1618922474.887855069]: called waypointV2UploadMissionCallback
[4014771.597]ERRORLOG/1 @ uploadWaypointV2, L2211: Upload waypoint v2 mission ErrorCode:0x110000010038

[4014771.597]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : WaypointV2
[4014771.597]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : waypoint position exceed radius limit 
[4014771.597]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : none
[ INFO] [1618922474.897996742]: Set avoid function callback
[ INFO] [1618922474.910555464]: Set upwards avoid function callback
[ INFO] [1618922474.932961733]: called waypointV2StartMissionCallback
[4014771.638]ERRORLOG/1 @ startWaypointV2Mission, L2275: Start waypoint v2 mission ErrorCode:0x110000010036

[4014771.639]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : WaypointV2
[4014771.639]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : Unknown
[4014771.639]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : �bK��

仅改动了Onboard-SDK-ROS/src/dji_osdk_ros/samples/waypointV2_node.cpp代码中的经纬度，改动如下
———————————————————————————————————————————————————————
std::vector<dji_osdk_ros::WaypointV2> generatePolygonWaypoints(ros::NodeHandle &nh, float32_t radius, uint16_t polygonNum)
{
  // Let's create a vector to store our waypoints in.
  std::vector<dji_osdk_ros::WaypointV2> waypointList;
  dji_osdk_ros::WaypointV2 startPoint;
  dji_osdk_ros::WaypointV2 waypointV2;

  startPoint.latitude  = gps_position_.latitude * C_PI / 180.0;
  startPoint.longitude = gps_position_.longitude * C_PI / 180.0;
  startPoint.relativeHeight = 15;
  setWaypointV2Defaults(startPoint);
  waypointList.push_back(startPoint);

  // Iterative algorithm
  for (int i = 0; i < polygonNum; i++) {
    float32_t angle = i * 2 * M_PI / polygonNum;
    setWaypointV2Defaults(waypointV2);
    float32_t X = radius * cos(angle);
    float32_t Y = radius * sin(angle);
    waypointV2.latitude = startPoint.latitude + 0.00001 * (i % 2);//Y/EARTH_RADIUS + startPoint.latitude;
    waypointV2.longitude = startPoint.longitude + 0.00001 * (i % 3);//X/(EARTH_RADIUS * cos(startPoint.latitude)) + startPoint.longitude;
    waypointV2.relativeHeight = startPoint.relativeHeight ;
    waypointList.push_back(waypointV2);
  }
  waypointList.push_back(startPoint);

  return waypointList;
}
——————————————————————————————————————————————————————————

dji-dev, 2021-04-21:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/46769'>Zendesk ticket #46769</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

建议您连接遥控器并打开APP后再尝试上传航点任务。
 [image]

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持​

guojing-gj, 2021-04-21:
连接上遥控器后解决了该问题，但是在输入的经纬度为以下数据时出现报错
lon， lat， alt
112.9667902, 27.84608351, 42.84326172
112.9667918, 27.84637506, 42.77197266
112.9663652, 27.84637687, 42.700737
112.9659386, 27.84637867, 42.62950897
112.965512, 27.84638047, 42.55827332
112.9655104, 27.84608892, 42.62955475
112.9655088, 27.84579736, 42.70085144
112.9655088, 27.84579736, 42.70085144
112.9659354, 27.84579556, 42.7720871
112.9663621, 27.84579375, 42.84331512
112.9667887, 27.84579195, 42.91455078

[ INFO] [1618987004.328329294]: called waypointV2UploadMissionCallback
[4079301.038]ERRORLOG/1 @ uploadWaypointV2, L2211: Upload waypoint v2 mission ErrorCode:0x11000001000A

[4079301.039]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : WaypointV2
[4079301.039]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : it is too close from two adjacent waypoints
[4079301.039]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : none
[ INFO] [1618987004.339030018]: Set avoid function callback
[ INFO] [1618987004.366306332]: Set upwards avoid function callback
[ INFO] [1618987004.387276671]: called waypointV2StartMissionCallback
[4079301.093]ERRORLOG/1 @ startWaypointV2Mission, L2275: Start waypoint v2 mission ErrorCode:0x110000010036

[4079301.093]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : WaypointV2
[4079301.093]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : Unknown
[4079301.093]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : @�&�