dji-sdk/Onboard-SDK-ROS #466: M600 pro demo failed

kimberz-git, 2021-05-23:
HI 
      my series is M600 pro, and i use N3 to connect it, the OSDK version is 3.9, 
when is start to run the demo at OSDK3.9, i always meet a problem when i start to send cmd to N3 to execute some operation, there is no feedback seen at DJI Assistant 2.
here is work log:

STM32F4Discovery Board initialization finished!
Sample App for STM32F4Discovery Board:

Prerequisites:
1. Vehicle connected to the Assistant and simulation is ON
2. Battery fully chanrged
3. DJIGO App connected (for the first time run)
4. Gimbal mounted if needed

STATUS/1 @ DJI::OSDK::Vehicle::parseDroneVersionInfo, L762: Device Serial No. = 0DBDJ14002002P

STATUS/1 @ DJI::OSDK::Vehicle::parseDroneVersionInfo, L764: Hardware = N3

STATUS/1 @ DJI::OSDK::Vehicle::parseDroneVersionInfo, L765: Firmware = 3.3.8.47

STATUS/1 @ DJI::OSDK::Vehicle::getDroneVersionCallback, L1626: Device Serial No. = 0DBDJ14002002P

STATUS/1 @ DJI::OSDK::Vehicle::getDroneVersionCallback, L1629: Hardware = N3

STATUS/1 @ DJI::OSDK::Vehicle::getDroneVersionCallback, L1630: Firmware = 303082F

STATUS/1 @ DJI::OSDK::DataSubscription::removePackage, L471: Remove package 0 successful.

ERRORLOG/1 @ DJI::OSDK::DataSubscription::removeLeftOverPackages, L503: failed to remove package 0

ERRORLOG/1 @ DJI::OSDK::Vehicle::initVirtualRC, L1188: Virtual RC is not supported on this platform!

STATUS/1 @ DJI::OSDK::Vehicle::parseDroneVersionInfo, L762: Device Serial No. = 0DBDJ14002002P

STATUS/1 @ DJI::OSDK::Vehicle::parseDroneVersionInfo, L764: Hardware = N3

STATUS/1 @ DJI::OSDK::Vehicle::parseDroneVersionInfo, L765: Firmware = 3.3.8.47

STATUS/1 @ DJI::OSDK::Vehicle::getDroneVersionCallback, L1626: Device Serial No. = 0DBDJ14002002P

STATUS/1 @ DJI::OSDK::Vehicle::getDroneVersionCallback, L1629: Hardware = N3

STATUS/1 @ DJI::OSDK::Vehicle::getDroneVersionCallback, L1630: Firmware = 303082F

ERRORLOG/1 @ DJI::OSDK::Vehicle::initVirtualRC, L1188: Virtual RC is not supported on this platform!

STATUS/1 @ DJI::OSDK::Vehicle::activate, L1361: version 0x303082F

STATUS/1 @ DJI::OSDK::ACK::getErrorCodeMessage, L752: 
DJI::OSDK::Vehicle::activateCallback

STATUS/1 @ DJI::OSDK::ACK::getCMDSetActivationMSG, L801: ACTIVATE_SUCCESS

STATUS/1 @ DJI::OSDK::DataSubscription::verifyCallback, L223: 
Verify subscription successful.

STATUS/1 @ DJI::OSDK::ACK::getErrorCodeMessage, L752: 
DJI::OSDK::Vehicle::controlAuthorityCallback

STATUS/1 @ DJI::OSDK::ACK::getCMDIDSetControlMSG, L899: OBTAIN_CONTROL_SUCCESS

Starting executing position control sample:

STATUS/1 @ DJI::OSDK::DataSubscription::addPackageCallback, L301: 
Start package 0 result: 0.

STATUS/1 @ DJI::OSDK::DataSubscription::addPackageCallback, L303: Package 0 info: freq=10, nTopics=2.
Takeoff failed. Motors failed to start!

STATUS/1 @ DJI::OSDK::DataSubscription::removePackageCallback, L443: Remove package 0 successful.
                                                                                                 STATUS/1 @ DJI::OSDK::DataSubscription::addPackageCallback, L301: 
Start package 0 result: 0.

STATUS/1 @ DJI::OSDK::DataSubscription::addPackageCallback, L303: Package 0 info: freq=50, nTopics=2.

STATUS/1 @ DJI::OSDK::ACK::getErrorCodeMessage, L752: 
DJI::OSDK::DataBroadcast::setFrequencyCallback,                             **is this a root cause for my problem ?** 

STATUS/1 @ DJI::OSDK::ACK::getCMDSetActivationMSG, L801: ACTIVATE_SUCCESS

dji-dev, 2021-05-24:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/48070'>Zendesk ticket #48070</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

If you don’t use OSDK, can you use the remote control to control the drone? If it works in DJI Assistant2 and the actual flight is wrong, it is recommended to use the remote control to ensure that the drone works.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

SD145, 2022-06-05:
CD/#CVN35023>FN(@)DEE^(((552K_)NUS2>>FM
# OpenJS Foundation and other contributors? | jquery.org/license//**SUSE#M tyre-CFR(***DS31[F]I30+3DFX76
# !function(e,t){"use strict";"object"==typeof module&Copy (HCS)

SD145, 2022-06-05:
Teld.com