dji-sdk/Onboard-SDK-ROS #468: M300 RTK与TX2连接问题

qiuminghilllittlewan, 2021-06-11:
我想用TX2 ROS订阅M300 fpv摄像头的话题数据，是需要把M300的osdk的拓展接口连接到TX2的USB上吗，如果是，需要怎么连接（好像没有OSDK->USB的线）

dji-dev, 2021-06-11:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/48881'>Zendesk ticket #48881</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

机载计算机与M300 通过串口和USB进行通信，OSDK拓展USB接口可以通过双A USB连接。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

qiuminghilllittlewan, 2021-06-11:
> Agent comment from kyle.cai in [Zendesk ticket #48881](https://djisdksupport.zendesk.com/agent/tickets/48881):
> 
> 尊敬的开发者，
> 
> 您好！感谢您联系DJI大疆创新。
> 
> 机载计算机与M300 通过串口和USB进行通信，OSDK拓展USB接口可以通过双A USB连接。
> 
> 感谢您的理解与支持,祝您生活愉快!
> 
> Best Regards，
> DJI大疆创新SDK技术支持
> 
> 
> _°°°_

所以我需要一个OSDK拓展USB的接口吗，有链接吗，谢谢