dji-sdk/Onboard-SDK-ROS #469: L1201: Start actions mode 15 failed, current DISPLAYMODE is: 12 ... (M210V2RTK)

Avistadrone, 2021-06-16:
Hello,

We are controlling a Matrice 210 RTK (lastest firmware) with the new interface 'dji_vehicle_node', to do so, flight_controler_node sample is used as a template, so we are just sending position offsets to the drone using MoveByPoseOffse and once the mission is finished we tried to send a go_home_and_Land command using the service flight_task_control and the task TASK_GOHOME_AND_CONFIRM_LANDING and the drone decided to land instead of going home throwing the error Start actions mode 15 failed, current DISPLAYMODE is: 12 ... It looks like the mode 15 is  MODE_NAVI_GO_HOME  and 12 is  MODE_AUTO_LANDING https://developer.dji.com/onboard-api-reference/namespaceDJI_1_1OSDK_1_1VehicleStatus_1_1DisplayMode.html
```

anuc@anuc:~/workspace/ros/aerostack_catkin_ws/src/aerostack_stack/launchers/dji_launchers/matrice_210$ roscore/master is 

ROS_MASTER_URI=http://anuc:11311

started roslaunch server http://anuc:32787/

SUMMARY
========

PARAMETERS
 * /drone6/vehicle_node/acm_name: /dev/ttyACM0
 * /drone6/vehicle_node/align_time: False
 * /drone6/vehicle_node/app_id: 1073520
 * /drone6/vehicle_node/app_version: 1
 * /drone6/vehicle_node/baud_rate: 921600
 * /drone6/vehicle_node/enc_key: 1ce9987d181096cb7...
 * /drone6/vehicle_node/serial_name: /dev/ttyUSB0
 * /drone6/vehicle_node/use_broadcast: True
 * /rosdistro: kinetic
 * /rosversion: 1.12.16

NODES
  /drone6/
    vehicle_node (dji_osdk_ros/dji_vehicle_node)

[330068.080]ERRORLOG/1 @ initVehicle, L161: Failed to initialize ACM Linker channel!
process[drone6/link1_broadcaster-2]: started with pid [18497]
process[drone6/link2_broadcaster-3]: started with pid [18518]
[330068.434]STATUS/1 @ getDroneVersion, L1702: ret = 0
[330068.434]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1DADFCQ001000T
[330068.434]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.3.44
[330068.434]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[330068.434]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[ INFO] [1623827706.747388234]: Inserting: /drone6/read_mission_points with state: ReadyToStart
[330069.434]STATUS/1 @ sendHeartbeatToFCTask, L1576: OSDK send heart beat to fc task created.
[330069.634]STATUS/1 @ Control, L40: The control class is going to be deprecated.It will be better to use the FlightController class instead!
[330069.634]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[330069.635]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[330069.635]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[330069.635]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[330069.635]STATUS/1 @ initDJIHms, L900: DJI HMS is not supported on this platform!

[330069.635]STATUS/1 @ init, L247: USB is not plugged or initialized successfully. Advacned-Sensing will not run.
[330069.635]STATUS/1 @ activate, L1329: version 0x304032C

[330069.653]STATUS/1 @ activate, L1367: Activation successful

[ INFO] [1623827708.542600679]: VehicleNode Start
[ INFO] [1623827708.542868256]: Topic startup!
[ INFO] [1623827708.711095892]: Services startup!
[ INFO] [1623827708.736053818]: Use legacy data broadcast to get telemetry data!
[330070.873]STATUS/1 @ verify, L244: Verify subscription successful.
[330070.893]STATUS/1 @ startPackage, L345: Start package 1 result: 0.
[330070.893]STATUS/1 @ startPackage, L347: Package 1 info: freq=50, nTopics=6.

[ INFO] [1623827724.141388650]: get Whole Battery Info callback
[ INFO] [1623827724.645092453]: Currrent GPS health is 5
[ INFO] [1623827724.645224755]: Local Position reference has been set.
[ INFO] [1623827724.645317241]: MONITOR GPS HEALTH WHEN USING THIS TOPIC
[ INFO] [1623827724.648968400]: Set current aircraft location as new home point callback
[330086.774]STATUS/1 @ verify, L244: Verify subscription successful.
[330086.793]STATUS/1 @ startPackage, L345: Start package 0 result: 0.
[330086.793]STATUS/1 @ startPackage, L347: Package 0 info: freq=1, nTopics=2.
[330088.813]STATUS/1 @ removePackage, L457: Remove package 0 successful.
[330088.832]STATUS/1 @ setCurrentAircraftLocAsHomePoint, L1375: Set new home location successfully
[ INFO] [1623827726.725388980]: Obtain Control Authority Callback
obtained control 
[ INFO] [1623827726.764523812]: Takeoff request sending ...
[ INFO] [1623827726.769199909]: call takeoff service
Motors spinning...
Ascending...
Successful takeoff!
[ INFO] [1623827733.287435859]: Takeoff task successful
Sending the following waypoint: -0.11 / -0.085 / 6.035 / -0
[ INFO] [1623827735.291367418]: call move local position offset service
Sending the following waypoint: -0.00146019 / 0.998872 / 0.142708 / 1.02264e-07
[ INFO] [1623827747.505571390]: call move local position offset service
Sending the following waypoint: 14.9996 / -0.974267 / 0.00421825 / -1.10872e-07
[ INFO] [1623827752.925600143]: call move local position offset service
Sending the following waypoint: 0.0844275 / 9.99123 / 0.992577 / 7.18516e-07
[ INFO] [1623827772.242199553]: call move local position offset service
Sending the following waypoint: -12.0079 / -5.91753 / -1.99787 / -3.82481e-07
[ INFO] [1623827786.765474195]: call move local position offset service
[ INFO] [1623827802.985153256]: call go home and confirm landing service
[330165.115]STATUS/1 @ verify, L244: Verify subscription successful.
[330165.134]STATUS/1 @ startPackage, L345: Start package 0 result: 0.
[330165.134]STATUS/1 @ startPackage, L347: Package 0 info: freq=10, nTopics=1.

[330165.134]STATUS/1 @ goHomeAndConfirmLanding, L1227: Start go home action
[330167.162]ERRORLOG/1 @ checkActionStarted, L1201: Start actions mode 15 failed, current DISPLAYMODE is: 12 ...
[ INFO] [1623827805.051004589]: Land task successful
Finished the mission
Smart-inspection photo counter is 0

```
The task is called as follows:
```
control_task_.request.task = dji_osdk_ros::FlightTaskControl::Request::TASK_GOHOME_AND_CONFIRM_LANDING;
task_flight_control_client_.call(control_task_);
if(control_task_.response.result == true){ROS_INFO_STREAM("Land task successful");}

```

lorenzmhe, 2021-06-16:
I suppose this could happen because the drone is within the threshold radius from the homepoint where it will land at its current position. I believe it is 15 or 25 meters but you can check that in the manual. You can also check the behavior of pressing RTH on the smart controller which would also land the drone at the current position if within the threshold radius.

Avistadrone, 2021-06-16:
@lorenzmhe you were right, but I wasn't expecting this kind of behaviour using the Onboard SDK, I guess I will have to implement the RTH logic by myself. 

Thank you very much.

@dji-dev any solution to this using the Onboard SDK?

Thanks in advance.

lorenzmhe, 2021-06-16:
You are very welcome. We discovered this feature by coincidence as well.

dji-dev, 2021-06-17:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/49037'>Zendesk ticket #49037</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

When the current point is less than 20m away from the home point, the landing will be executed directly while calling the API to return home.
Developers can impelement this part with OSDK flightCtrl features.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

closed 2021-06-17 by Avistadrone