dji-sdk/Onboard-SDK-ROS #470: ./MatriceSeries_ConfigTool --usb-port /dev/ttyACM0 --c./MatriceSeries_ConfigTool: error while loading shared libraries: libavcodec.so.57: cannot open shared object file: No such file or directory

Avistadrone, 2021-06-22:
Hello,

I already opened this issue in the Onboard SDK repository, but I am opening here in case anyone can help me.
I have been trying to activate the flag that let the drone fly with the USB A- USB A cable connected but I am getting the following error:

./MatriceSeries_ConfigTool --usb-port /dev/ttyACM0 --config-file /home/anuc/.ros/UserConfig.txt --usb-connected-flight on
./MatriceSeries_ConfigTool: error while loading shared libraries: libavcodec.so.57: cannot open shared object file: No such file or directory

Everything in the SDK work nice but the MatriceSreis_ConfigTool, it seems like the only package that depends on libavcodec.57.

I am running this in an intel Nuc I7 with Ubuntu 16.04.
```
dpkg -l "*avcodec*" | grep ii
ii  libavcodec-dev:amd64            7:2.8.17-0ubuntu0.1 amd64        FFmpeg library with de/encoders for audio/video codecs - development files
ii  libavcodec-ffmpeg-extra56:amd64 7:2.8.17-0ubuntu0.1 amd64        FFmpeg library with additional de/encoders for audio/video codecs
```

I followed this link: https://developer.dji.com/onboard-sdk/documentation/quickstart/development-environment.html

And when ffmpeg is installed as you said in the tutorial I get that the library is the newest version available for Ubuntu 16.04, which is the version in which you have done all the tutorial...

Install Development Tools

C ++ compiler: GCC 5.4.0/5.5.0 version
CMake: 2.8 and above
Linux: Ubuntu 16.04 (If you need to use advanced sensor functions, please use the Libusb library)
```
sudo apt-get install libavcodec-dev libswresample-dev
[sudo] password for anuc: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libavcodec-dev is already the newest version (7:2.8.17-0ubuntu0.1).
libswresample-dev is already the newest version (7:2.8.17-0ubuntu0.1).
The following packages were automatically installed and are no longer required:
  libgavl1 libhdf5-mpi-dev libhdf5-openmpi-10 libhdf5-openmpi-dev libignition-cmake-dev libignition-math4
  libignition-math4-dev libignition-msgs libignition-msgs-dev libignition-transport4 libignition-transport4-dev
  libqwt-headers libqwt-qt5-6 libqwt-qt5-dev libyaml-dev libzip-dev libzip4 libzmq3-dev
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 112 not upgraded.
```

dji-dev, 2021-06-22:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/49265'>Zendesk ticket #49265</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

I am sorry that because of the email system, I have not received the updates about the Onboard SDK repository issue #792, so I ignored the issue updates. If need DJI technical support ,we suggest you send email to dev@dji.com.

About your issue, there is no intel Nuc I7 to test, and have you tried to run the MatriceSeries_ConfigTool  on another platform?
I test it on PC Vmware and Manifold 2G, but can't reproduce the issue. suggest that you can change the platform for ensure the SDK works and then to check the compiling and running enviroment on your own onboard computer.
 [image]​​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/N1DzPJETtPzBp5cIrNpcwhZVi' target='zd_attach'>inline-619264724.png</a>
<br><p><em><small>°°°</small></em></p>

Avistadrone, 2021-06-22:
Hello,

Thank you for the answer.
I am afraid we only work with Intel Nucs, I guess I will try updating to ROS Melodic since libavcoded.57 can be installed in that ubuntu's version. But honestly, I don't understand how you can run the ./MatriceSeries_ConfigTool with that libavcodec version.

Could you please run the script and let me know the result?

As I told you before, everything in the SDK works good but the MatriceSeries_ConfigTool since it apparently depends on a higher version of libavcodec.

Thank you.

dji-dev, 2021-06-23:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/49265'>Zendesk ticket #49265</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

I test it with x86 vmware ubuntu 1604
 [image]

​ [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/NtJv4goIrhzcoIT8bEhQkTVKM' target='zd_attach'>inline1363097537.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/n8yfDitVDgEykuLhrXjzvyKkR' target='zd_attach'>inline1534669010.png</a>
<br><p><em><small>°°°</small></em></p>

lorenzmhe, 2021-06-23:
If this keeps challenging you, the source is [here](https://github.com/dji-sdk/Onboard-SDK/tree/master/sample/platform/linux/config_tool) so I image that you can remove the libav part (although I don't see it by quick inspection).

Avistadrone, 2021-06-23:
@lorenzmhe Thank you for your help, I could't see it either so I decided to upgrade to ubuntu 18.04 and it seems to work fine.

closed 2021-06-23 by Avistadrone