dji-sdk/Onboard-SDK-ROS #471: How does "failsafe" (low battery, NFZ, etc.) work with OSDK on M300

lorenzmhe, 2021-07-05:
Please explain how the drone reacts/should react when:

- An onboard computer has control authority and the battery gets low
- An onboard computer attempts to perform a takeoff in a No Flight Zone
- An onboard computer commands the drone to enter a No Flight Zone

Previously, the drone would always be in control and stop when entering NFZ and not be able to be controlled by OSDK in NFZ. Today, we experience this to be less deterministic and we need to know what we can depend on.

dji-dev, 2021-07-06:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/49866'>Zendesk ticket #49866</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

- An onboard computer has control authority and the battery gets low

-->If the battery does not support takeoff, OSDK will report an error and the drone can't taken off. During the flight, the control will be executed according to the set by the APP (such as low battery RTH)

- An onboard computer attempts to perform a takeoff in a No Flight Zone
- An onboard computer commands the drone to enter a No Flight Zone
     -->Cannot take off in the no-fly zone, it will hover when encountering the no-fly zone during the flight

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

lorenzmhe, 2021-07-06:
Thank you for the quick response.

lorenzmhe, 2021-07-06:
How is the behavior if RC is lost during flight?