dji-sdk/Onboard-SDK-ROS #472: Waypoint Mission does not start m600

Oburshan, 2021-07-11:
Hey, I am currently writing a simple waypoint ROS node that converts **path** massage into an official DJI waypoint task.
DJI model - m600, branch 3.8.

it calls 3 services: **sdk_control_authority**, **mission_waypoint_upload** and **mission_waypoint_action**

It seems that everything works fine but nothing happens inside the simulation.. this is the log I got : 

[ INFO] [1626015343.213063364]: drone activated
[ INFO] [1626015343.351111936]: Use legacy data broadcast to get telemetry data!

STATUS/1 @ getErrorCodeMessage, L656: droneTaskCallback
STATUS/1 @ getCommonErrorCodeMessage, L726: UNKNOWN_ACK_ERROR_CODE
[ INFO] [1626015740.636652713]: initialized waypoint mission
[ INFO] [1626015741.656410954]: uploaded the 1th waypoint

[ INFO] [1626015742.679364011]: uploaded the 2th waypoint

[ INFO] [1626015743.679618718]: waypoint mission initialized and uploaded

STATUS/1 @ getErrorCodeMessage, L656: missionWpActionCallback
STATUS/1 @ getCMDSetMissionMSG, L883: MISSION_IN_PROGRESS

dji-dev, 2021-07-12:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/50151'>Zendesk ticket #50151</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Does the drone works if you using the original sample ? Please make sure the RC and APP has been connected, if the radius exceed 50m, and hight exceed 30m.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

Oburshan, 2021-07-12:
Hey,
The RC and App are connected and the demo node works fine...
Should I call other services besides the 3 I mentioned?

Oburshan, 2021-07-18:
Solved, I called 1 instead of 0 in the mission waypoint action.

closed 2021-07-18 by Oburshan