dji-sdk/Onboard-SDK-ROS #473: How to properly navigate using the RTK Data

Avistadrone, 2021-07-14:
Hello,

We are currently trying to perform an autonomous flight using the MoveByPositionOffset method, however we have noticed that the globalPosition does not use the RTK data, so that the pose estimation is not that accurate. In order to achieve a better accuracy, we modified the method MoveByPositionOffset so that it takes the gps location from the topic RTK_POSITION, but its frequency it's just 5Hz what it's quite poor.
What would be the best approach to achieve a gps autonomous flight using the rtk information? Using the mission control? As far as we know, the mission control doesn't let you control camera's pan, that's why we tried with the MoveByPositionOffset method.

Thanks.

Best regards.

dji-dev, 2021-07-15:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/50312'>Zendesk ticket #50312</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

For mission control, waypoint actions can be set to control gimbal while reaching the waypoint.
Unfortunately, if you use MoveByPositionOffset and get RTK_POSITION through FC, the highest frequency of 5HZ can't be improved.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

Avistadrone, 2021-07-15:
Thank you for the answer.

Could you let us know how to control the yaw of the gimbal using the waypoint action because there is no such attribute in the struct WayPointSettings?
To make everything clear, if the RTK is connected, the waypoints mission will automatically fuse the RTK data and obtain a global Position, right? In case it does, there is no way to just subscribe to the fused estimated pose using ROS topics? 

Thanks.

Best regards.

ahmed-elsaharti, 2021-08-11:
@Avistadrone WayPointSettings does seem to have a yaw attribute https://github.com/dji-sdk/Onboard-SDK/blob/dc4b77d48fbc8af47ab95e05cc237e0212e9413f/osdk-core/api/inc/dji_mission_type.hpp#L133