dji-sdk/Onboard-SDK-ROS #474: 关于如何室内控制飞行？（如何使用 attitude_control API）

Konye666, 2021-08-11:
您好，通过查阅[wiki](http://wiki.ros.org/dji_sdk/)以及[部分答复](https://github.com/dji-sdk/Onboard-SDK-ROS/issues/449)，已知在定位环境下能够通过基于 flight_control 接口实现的话题类型 /dji_sdk/flight_control_setpoint_generic 等一系列话题进行控制。

但在室内环境（或不依靠定位信息）的情况下，按以上答复以及网上部分资料，需要使用 attitude_control 接口进行控制，请问还有其它控制的形式或者接口吗？

以及关于该接口，网上提供的一些指向开发者网页的地址已经失效，请问现在能在哪里获取相关资料？在Github上搜索attitude_control字眼也未能找到对应源代码部分。谢谢

dji-dev, 2021-08-12:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/51602'>Zendesk ticket #51602</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

DJI无人机本身不提供室内无GPS方案，若使用OSDK控制，仅支持使用flightCtrl姿态模式进行控制，定位或导航均需开发者自行开发。
OSDK飞行控制和模式定义，可以参考论坛相关帖：
https://bbs.dji.com/forum.php?mod=viewthread&tid=258561

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

Konye666, 2021-08-12:
感谢回复。
粗略浏览您提供的的相关贴，已成功在OSDK源码库中寻找到相关接口代码。根据帖子所提供的信息，请问姿态模式指的是以机体坐标系FRU为依照的角度、速度、位置等一些列控制吗？不知我的理解有误，在此模式下，角度、速度、位置是依靠飞机自身飞控上的传感进行校准吗（暂不考虑由开发者自行开发引入的第三方定位信息或算法）？
谢谢。