dji-sdk/Onboard-SDK-ROS #477: M600 [dji_sdk-2] process has died

alduxvm, 2021-09-18:
Hi!

I have an M600 with an onboard computer attach to it, the computer is running ubuntu 18.04, ROS Melodic. I'm currently using tag 3.8.1 of the Onboard-SDK-ROS, compilation work great, I disabled advaced_sensing, I have the proper app id and key, when launching the node, it connects to the drone but it crashes some seconds after starting... the output looks as follows:

```
$ roslaunch dji_sdk sdk.launch 
... logging to /home/aldux/.ros/log/8c46c216-18bc-11ec-8ca5-50eb71403eef/roslaunch-drone-13163.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://drone:39291/

SUMMARY
========

PARAMETERS
 * /dji_sdk/acm_name: /dev/ttyACM0
 * /dji_sdk/align_time: False
 * /dji_sdk/app_id: 1108879
 * /dji_sdk/app_version: 1
 * /dji_sdk/baud_rate: 921600
 * /dji_sdk/enc_key: XXXX...
 * /dji_sdk/serial_name: /dev/ttyUSB0
 * /dji_sdk/use_broadcast: False
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    dji_sdk (dji_sdk/dji_sdk_node)

auto-starting new master
process[master]: started with pid [13173]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 8c46c216-18bc-11ec-8ca5-50eb71403eef
process[rosout-1]: started with pid [13184]
started core service [/rosout]
process[dji_sdk-2]: started with pid [13187]

STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 921600...

STATUS/1 @ init, L65: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L727: Device Serial No. = XXXXXX

STATUS/1 @ parseDroneVersionInfo, L729: Hardware = PM820V3PRO

STATUS/1 @ parseDroneVersionInfo, L730: Firmware = 3.2.41.14

ERRORLOG/1 @ initVirtualRC, L1103: Virtual RC is not supported on this platform!

STATUS/1 @ activate, L1313: version 0x302290E

STATUS/1 @ activate, L1326: Activation successful

STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ initGimbal, L890: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=50, nTopics=1.

STATUS/1 @ removePackage, L468: Remove package 0 successful.[ INFO] [1631995870.001634335]: drone activated
[dji_sdk-2] process has died [pid 13187, exit code -11, cmd /home/aldux/m600_ros/devel/lib/dji_sdk/dji_sdk_node __name:=dji_sdk __log:=/home/aldux/.ros/log/8c46c216-18bc-11ec-8ca5-50eb71403eef/dji_sdk-2.log].
log file: /home/aldux/.ros/log/8c46c216-18bc-11ec-8ca5-50eb71403eef/dji_sdk-2*.log
```

I have several questions:

1.- How can I prevent this crash? 
2.- Is tag 3.8.1 the last tag that can be used for the m600?

Thanks a lot!!

dji-dev, 2021-09-22:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/53552'>Zendesk ticket #53552</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

1、The OSDK（Linux）3.9 and OSDK-ROS 3.8 is recommended for M600.

2、About the crash, is there any code changes? I test it with the orignal demo, it works. if it is convient, please share your code.
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/q3Zr3B2cSG5OVPROMua9fWovj' target='zd_attach'>inline618213517.png</a>
<br><p><em><small>°°°</small></em></p>

alduxvm, 2021-09-22:
Hi!

Thanks for the answer! 

I did not modified anything from the code, I repeated my test using OSDK 3.9 (I was using 4.0) and OSDK-ROS 3.8.1... and now I get the next memory error:

```
ERRORLOG/1 @ getValue, L315: Topic 0x26 value memory not initialized, return default
```

Googling that error I found the next issue in [github](https://github.com/dji-sdk/Onboard-SDK-ROS/issues/307) and I tested one of the solutions from the DJI tech people, where they attached a .tar.gz file, I download it on my computer, compile and then I tested and its working! now I can see the topics! 

My conclusion is that the current committed code for ROS 3.8.1 seems to be not working.

dji-dev, 2021-09-23:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/53552'>Zendesk ticket #53552</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

OK, please allow me to explain about the error:

    ERRORLOG/1 @ getValue, L315: Topic 0x26 value memory not initialized, return default

It's because M600 doesn't support TOPIC: TOPIC_RTK_CONNECT_STATUS, it hasn't been initialized. If you are using RTK, please remove TOPIC.
 [image]​
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/9nzve1zKNcfcxfdQRoF9UXs0h' target='zd_attach'>inline-1929827045.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/W3sSoqZ2Q76JKrnsss03JH5b1' target='zd_attach'>inline740491008.png</a>
<br><p><em><small>°°°</small></em></p>

Souyu, 2021-11-02:
I'm also getting the same error.M600 Pro, running OSDK 3.9 and OSDK ROS 3.8.1 Do I need to connect a gimbal in order for it to work?

```
... logging to /home/dji/.ros/log/fb9575d4-3b8c-11ec-a122-60601fcd6335/roslaunch-manifold2-11309.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://manifold2:44930/

SUMMARY
========

PARAMETERS
 * /dji_sdk/acm_name: /dev/ttyACM0
 * /dji_sdk/align_time: False
 * /dji_sdk/app_id: 1110951
 * /dji_sdk/app_version: 1
 * /dji_sdk/baud_rate: 921600
 * /dji_sdk/enc_key: cd19ddc21d80dd04b...
 * /dji_sdk/serial_name: /dev/ttyTHS2
 * /dji_sdk/use_broadcast: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.17

NODES
  /
    dji_sdk (dji_sdk/dji_sdk_node)

ROS_MASTER_URI=http://localhost:11311

process[dji_sdk-1]: started with pid [11413]
[ INFO] [1635824893.424183828]: Advanced Sensing is Enabled on M210.

STATUS/1 @ init, L55: Attempting to open device /dev/ttyTHS2 with baudrate 921600...

STATUS/1 @ init, L65: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L727: Device Serial No. = 0670164661

STATUS/1 @ parseDroneVersionInfo, L729: Hardware = PM820V3PRO

STATUS/1 @ parseDroneVersionInfo, L730: Firmware = 3.2.41.14

ERRORLOG/1 @ initVirtualRC, L1103: Virtual RC is not supported on this platform!

STATUS/1 @ activate, L1313: version 0x302290E

STATUS/1 @ activate, L1326: Activation successful

STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ initGimbal, L890: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=50, nTopics=1.

STATUS/1 @ removePackage, L468: Remove package 0 successful.
STATUS/1 @ initGimbal, L934: Gimbal not mounted!
[ INFO] [1635824899.736292778]: drone activated
[dji_sdk-1] process has died [pid 11413, exit code -11, cmd /home/dji/Desktop/catkin_ws_3.8/devel/lib/dji_sdk/dji_sdk_node __name:=dji_sdk __log:=/home/dji/.ros/log/fb9575d4-3b8c-11ec-a122-60601fcd6335/dji_sdk-1.log].
log file: /home/dji/.ros/log/fb9575d4-3b8c-11ec-a122-60601fcd6335/dji_sdk-1*.log
```

alduxvm, 2021-11-02:
Hi Souyu!

Try the catkin from here: https://djisdksupport.zendesk.com/attachments/token/ZMwuOakcqm8GWSsarqprZwKmu

Thats how I got mine working!

Good luck!

babakbch, 2024-05-07:
Hi alduxvm
I am having the same problem with my dji M600 pro , I can't access the link you have shared in the last msg, so i can't see the solution. would you be so kind and share the solution? I am running dji-osdk and dji-osdk-ros both 3.8.1

alduxvm, 2024-05-08:
I have a copy, it took me a while to find, but here we go! you might need to recompile and put your ID as well... let me know if it works.
[catkin_600.zip](https://github.com/dji-sdk/Onboard-SDK-ROS/files/15250578/catkin_600.zip)