dji-sdk/Onboard-SDK-ROS #479: Initiate SDK node problem

dregner, 2021-10-04:
Hello,

I am trying to connect M210 v2 RTK to a desktop computer with Ubuntu 18.04, ROS Melodic and parallel installation of Opencv 3.3.1 and 4.5.3 using a USB-TTL RS232 to make UART connection and an USB-USB connecting drone and desktop to be able to run Advanced Sensing.

When I call `ls -l /dev/ttyACM* && ls -l /dev/ttyUSB*` it returns that it is indentified the USB and ACM connection.
```
crw-rw---- 1 root dialout 166, 0 out  4 13:18 /dev/ttyACM0
crw-rw---- 1 root dialout 188, 0 out  4 13:18 /dev/ttyUSB0
```

I also set the transfer rate of TTL-USB to 921600 using `minicom`, and gave persmission to device to read and write with `sudo usermod -a -G dialout $USER && sudo chmod 666 /dev/ttyUSB0` 

Unfortunatelly when I launch `roslaunch dji_osdk_ros dji_sdk_node.launch` it appears some connection problem presented below and I am not being able to fix it. I have been trying to turn on/off drone and RC several times ass described [here](https://forum.dji.com/thread-216529-1-1.html), but the problem still stand.

```
started roslaunch server http://V3D06:43613/

SUMMARY
========

PARAMETERS
 * /dji_sdk/acm_name: /dev/ttyACM0
 * /dji_sdk/align_time: False
 * /dji_sdk/app_id: 1076017
 * /dji_sdk/app_version: 1
 * /dji_sdk/baud_rate: 921600
 * /dji_sdk/dxc: False
 * /dji_sdk/enc_key: 6bd1d26f8dd897e4b...
 * /dji_sdk/serial_name: /dev/ttyUSB0
 * /dji_sdk/use_broadcast: False
 * /rosdistro: melodic
 * /rosversion: 1.14.12

NODES
  /
    dji_sdk (dji_osdk_ros/dji_sdk_node)

auto-starting new master
process[master]: started with pid [2436]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to bde7b4d2-252e-11ec-8a59-1831bfb3e154
process[rosout-1]: started with pid [2458]
started core service [/rosout]
process[dji_sdk-2]: started with pid [2464]
[ INFO] [1633364323.534426789]: Advanced Sensing is Enabled on M210.
Read App ID
User Configuration read successfully. 

[1276751.089]STATUS/1 @ getDroneVersion, L1702: ret = 0
[1276751.089]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1DADG3E00100U4
[1276751.089]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.3.44
[1276751.089]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[1276751.089]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[1276752.089]STATUS/1 @ sendHeartbeatToFCTask, L1576: OSDK send heart beat to fc task created.
[1276752.289]STATUS/1 @ Control, L40: The control class is going to be deprecated.It will be better to use the FlightController class instead!
[1276752.290]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[1276753.557]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[1276753.557]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[1276753.557]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[1276753.557]STATUS/1 @ initDJIHms, L900: DJI HMS is not supported on this platform!

[1276753.567]STATUS/1 @ getDroneVersion, L1702: ret = 0
[1276753.567]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1DADG3E00100U4
[1276753.567]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.3.44
[1276753.567]STATUS/1 @ AdvancedSensing, L145: Advanced Sensing init for the M210 drone
[1276753.567]STATUS/1 @ init, L49: Looking for USB device...

[1276753.572]STATUS/1 @ init, L65: Found 8 USB devices, identifying DJI device...

[1276753.572]STATUS/1 @ init, L83: Found a DJI device...

[1276753.572]STATUS/1 @ init, L96: Attempting to open DJI USB device...

[1276753.572]ERRORLOG/1 @ init, L101: Failed to open DJI USB device...

[1276753.572]ERRORLOG/1 @ init, L102: Error code: -3
[1276753.572]ERRORLOG/1 @ init, L105: Please make sure you provide a udev file for your system and reboot the computer
[1276753.573]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.
[1276754.076]STATUS/1 @ init, L254: Start advanced sensing initalization
[1276754.076]STATUS/1 @ activate, L1329: version 0x304032C

[1276754.076]STATUS/1 @ adv_pthread, L46: adv pthread created !!!!!!!!!!!!!!!!!!!!!!!
[1276754.076]STATUS/1 @ adv_pthread, L48: adv pthread running !!!!!!!!!!!!!!!!!!!!!!!
[dji_sdk-2] process has died [pid 2464, exit code -11, cmd /home/vant3d/catkin_ws/devel/lib/dji_osdk_ros/dji_sdk_node __name:=dji_sdk __log:=/home/vant3d/.ros/log/bde7b4d2-252e-11ec-8a59-1831bfb3e154/dji_sdk-2.log].
log file: /home/vant3d/.ros/log/bde7b4d2-252e-11ec-8a59-1831bfb3e154/dji_sdk-2*.log
```

It appears it has some problem providing a udev file, but I don't know how to fix it. Does anyone have some idea to help on this problems? 

Thank you!

dji-dev, 2021-10-05:
Agent comment from William Wong in <a href='https://djisdksupport.zendesk.com/agent/tickets/54112'>Zendesk ticket #54112</a>:<br><br>Hi Developer,

Thank you for reach out to DJI Developer Support.  Due to a Chinese Holiday there maybe a short delay in responding directly to your question - we apologize for any inconvenience this may cause you and look forward to addressing your ticket as soon as possible.

Warm Regards,
DJI Developer Support
<br><p><em><small>°°°</small></em></p>

dji-dev, 2021-10-06:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/54112'>Zendesk ticket #54112</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Please try to run with “sudo roslaunch dji_osdk_ros dji_vehicle_node.launch”,
if it works, you can refer to the following to add DJI USB device node.
 [image]​
（For M210 v2 series drones, dji_vehicle_node is recommended, If the error persists, please provide the terminal log of "sudo roslaunch dji_osdk_ros dji_vehicle_node.launch"）.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/yIcnRyyY24uzpShOx7mBKgKL4' target='zd_attach'>inline-626522359.png</a>
<br><p><em><small>°°°</small></em></p>

dregner, 2021-10-14:
A couple days I was still facing this problem, I ended on this [thread](https://forum.dji.com/thread-222973-1-1.html) explaining the small mistake on yours documentation. After remove the following problem all went just fine. Can you please fix it?

> The command there is: **SUBSYSTEM=="usb", ATTRS {idVendor} =="2ca3", MODE ="0666"**
> However, the correct command is: **SUBSYSTEM=="usb", ATTRS{idVendor} =="2ca3", MODE ="0666"**
> 
> So, the space between ATTRS and {idVendor} is wrong. Leaving this here in case someone else runs into that problem.
Thank you