dji-sdk/Onboard-SDK-ROS #482: M100 run sample and get errors

s40523139, 2021-11-01:
**Hi everyone, I am a novice in using and developing drone.I encounter some error when I try to run the sample.
Here is my hardware setup.**
**I also open DJI Assistant2 while running the sample(as you can see in picture1)**
[image]                      [image]
[image]

**For the yellow line I have try several SDK version and get same error message (3.1、3.4、3.6、3.8.1)
content of Userconfig.txt  is writing like this
app_id : xxx
app_key : xxx
device :  /dev/ttyUSB0
baudrate : 230400**

[image]

**Can anyone give me some advice to solve this?**

dji-dev, 2021-11-01:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/55494'>Zendesk ticket #55494</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The error log seems that the version you are using is 4.x, please remove the OSDK first(including the lib), and download again from the following link:
https://github.com/dji-sdk/Onboard-SDK/tree/3.9

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

s40523139, 2021-11-01:
@dji-dev 
Thanks for replying.

But in the terminal I use version 3.8.1 to test the sample.

I have one question, you say remove the OSDK first(including the lib), what is the lib I should remove？Is it mean that I just only remove entire package (ex. OSDK 4.0) I clone from github？　

Thanks for your response!

dji-dev, 2021-11-01:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/55494'>Zendesk ticket #55494</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The error from the code:
master branch(OSDK 4.x)：
https://github.com/dji-sdk/Onboard-SDK/blob/d08abe61768f50486ff10bca991c524e6308a011/osdk-core/api/src/dji_vehicle.cpp#L298
 [image]

If you have install the lib(make install), please remove it from the install path.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team​<br><a href='https://djisdksupport.zendesk.com/attachments/token/pDR5QJjLmj0rsuTERsYhltsyH' target='zd_attach'>inline1122769938.png</a>
<br><p><em><small>°°°</small></em></p>

closed 2021-11-22 by s40523139