dji-sdk/Onboard-SDK-ROS #483: catkin_make error

fire717, 2021-11-02:
while I try to Build the dji_osdk_ros ROS package on Ubuntu18，
I run
catkin_make 
and got this:

`Base path: /home/yw/workshop/pipe/catkin_ws
Source space: /home/yw/workshop/pipe/catkin_ws/src
Build space: /home/yw/workshop/pipe/catkin_ws/build
Devel space: /home/yw/workshop/pipe/catkin_ws/devel
Install space: /home/yw/workshop/pipe/catkin_ws/install
####
#### Running command: "cmake /home/yw/workshop/pipe/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/yw/workshop/pipe/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/yw/workshop/pipe/catkin_ws/install -G Unix Makefiles" in "/home/yw/workshop/pipe/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/yw/workshop/pipe/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/yw/workshop/pipe/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 2 packages in topological order:
-- ~~  - nmea_msgs
-- ~~  - dji_osdk_ros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'nmea_msgs'
-- ==> add_subdirectory(nmea_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- nmea_msgs: 7 messages, 0 services
-- +++ processing catkin package: 'dji_osdk_ros'
-- ==> add_subdirectory(Onboard-SDK-ROS)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Support Advanced Sensing
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/x86_64-linux-gnu/libusb-1.0.so
-- dji_osdk_ros: 38 messages, 75 services
-- USB INCLUDE: /usr/include/libusb-1.0
-- Found OpenCV 4.5.0 installed in the system, will use it to display image in AdvancedSensing APIs
--  - Includes: /usr/local/include/opencv4
--  - Libraries: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_gapi;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_stitching;opencv_video;opencv_videoio
-- Found 10.2 CUDA installed in the system, will use it for depth perception sample
--  - Includes: /usr/local/cuda/include
-- PROJECT_NAME: dji_osdk_ros
-- DJIOSDK_LIB: 
-- Found SDL2:
--  - Includes: /usr/include/SDL2
--  - Libraries: -lSDL2
-- Find OPENCV, camera stream node will be compiled.
-- Did not find required libraries, stereo vision depth perception node will not be compiled.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yw/workshop/pipe/catkin_ws/build
####
#### Running command: "make -j16 -l16" in "/home/yw/workshop/pipe/catkin_ws/build"
####
Scanning dependencies of target _dji_osdk_ros_generate_messages_check_deps_SetupCameraH264
Scanning dependencies of target _dji_osdk_ros_generate_messages_check_deps_CameraTapZoomPoint
Scanning dependencies of target _dji_osdk_ros_generate_messages_check_deps_WaypointV2IntervalTrigger
Scanning dependencies of target _dji_osdk_ros_generate_messages_check_deps_MissionHpUpdateRadius
[... 880 lines cut ...]
make[2]: *** 正在等待未完成的任务....
[ 95%] Building C object Onboard-SDK-ROS/src/dji_osdk_ros_obsoleted/CMakeFiles/dji_sdk_node.dir/__/dji_osdk_ros/modules/osdkhal_linux.c.o
[ 95%] Building C object Onboard-SDK-ROS/src/dji_osdk_ros_obsoleted/CMakeFiles/dji_sdk_node.dir/__/dji_osdk_ros/modules/osdkosal_linux.c.o
[ 95%] Building CXX object Onboard-SDK-ROS/src/dji_osdk_ros_obsoleted/CMakeFiles/dji_sdk_node.dir/dji_linux_environment.cpp.o
[ 95%] Building CXX object Onboard-SDK-ROS/src/dji_osdk_ros_obsoleted/CMakeFiles/dji_sdk_node.dir/dji_linux_helpers.cpp.o
/home/yw/workshop/pipe/catkin_ws/src/Onboard-SDK-ROS/src/dji_osdk_ros/modules/dji_vehicle_node_mission_services.cpp: In member function ‘bool dji_osdk_ros::VehicleNode::waypointV2GenerateActionsCallback(dji_osdk_ros::GenerateWaypointV2Action::Request&, dji_osdk_ros::GenerateWaypointV2Action::Response&)’:
/home/yw/workshop/pipe/catkin_ws/src/Onboard-SDK-ROS/src/dji_osdk_ros/modules/dji_vehicle_node_mission_services.cpp:800:24: error: ‘DJI::OSDK::DJIWaypointV2CameraFocusParam {aka struct DJI::OSDK::DJIWaypointV2CameraFocusParam}’ has no member named ‘regionType’
             focusParam.regionType = request.actions[i].waypointV2CameraActuator.focusParam.regionType;
                        ^~~~~~~~~~
/home/yw/workshop/pipe/catkin_ws/src/Onboard-SDK-ROS/src/dji_osdk_ros/modules/dji_vehicle_node_mission_services.cpp:801:24: error: ‘DJI::OSDK::DJIWaypointV2CameraFocusParam {aka struct DJI::OSDK::DJIWaypointV2CameraFocusParam}’ has no member named ‘width’
             focusParam.width = request.actions[i].waypointV2CameraActuator.focusParam.width;
                        ^~~~~
/home/yw/workshop/pipe/catkin_ws/src/Onboard-SDK-ROS/src/dji_osdk_ros/modules/dji_vehicle_node_mission_services.cpp:802:24: error: ‘DJI::OSDK::DJIWaypointV2CameraFocusParam {aka struct DJI::OSDK::DJIWaypointV2CameraFocusParam}’ has no member named ‘height’
             focusParam.height = request.actions[i].waypointV2CameraActuator.focusParam.height;
                        ^~~~~~
Onboard-SDK-ROS/src/dji_osdk_ros/modules/CMakeFiles/dji_osdk_ros.dir/build.make:62: recipe for target 'Onboard-SDK-ROS/src/dji_osdk_ros/modules/CMakeFiles/dji_osdk_ros.dir/dji_vehicle_node_mission_services.cpp.o' failed
make[2]: *** [Onboard-SDK-ROS/src/dji_osdk_ros/modules/CMakeFiles/dji_osdk_ros.dir/dji_vehicle_node_mission_services.cpp.o] Error 1
CMakeFiles/Makefile2:7022: recipe for target 'Onboard-SDK-ROS/src/dji_osdk_ros/modules/CMakeFiles/dji_osdk_ros.dir/all' failed
make[1]: *** [Onboard-SDK-ROS/src/dji_osdk_ros/modules/CMakeFiles/dji_osdk_ros.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
[ 95%] Linking CXX executable /home/yw/workshop/pipe/catkin_ws/devel/lib/dji_osdk_ros/dji_sdk_node
[ 95%] Built target dji_sdk_node
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j16 -l16" failed`

dji-dev, 2021-11-02:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/55568'>Zendesk ticket #55568</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The OSDK ROS based on the OSDK linux, which version of OSDK linux are you using ? please check and comfirm the OSDK(linux) compatiable with ROS 3.8, and OSDK 4.1 with OSDK ROS 4.1.
 [image]

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team​<br><a href='https://djisdksupport.zendesk.com/attachments/token/2BAEo7rNi1khJR9dfZ1z7nF5P' target='zd_attach'>inline1269326892.png</a>
<br><p><em><small>°°°</small></em></p>

fire717, 2021-11-02:
解决了，是因为onboard sdk和onboard sdk ros版本不一致的问题

closed 2021-11-02 by fire717