dji-sdk/Onboard-SDK-ROS #484: where is the UserConfig.txt?

fire717, 2021-11-02:
I try to install this repo following the README
and I saw this:

> 3.Remember to add UserConfig.txt to correct path.(in the current work directory)

but I could not find any file named UserConfig.txt in Onboard-SDK-ROS or 
Onboard-SDK

so how I can I get it? what is writen in it?

dji-dev, 2021-11-02:
Agent comment from DJI SDK in <a href='https://djisdksupport.zendesk.com/agent/tickets/55594'>Zendesk ticket #55594</a>:<br><br>尊敬的开发者，感谢您联系DJI 大疆创新
由于github不是我们主要的咨询渠道，您的问题可能跟进不及时。我们建议您通过填写表单（ https://djisdksupport.zendesk.com/hc/zh-cn/requests/new ）向我们反馈问题。或者您也可以在论坛发帖，与其它开发者交流。论坛链接：https://djisdksupport.zendesk.com/hc/zh-cn/community/topics

Dear developer, thank you for contacting DJI.
Since github is not our main consultation channel, your questions may not be followed up in time. We recommend that you fill in the form (https://djisdksupport.zendesk.com/hc/en-us/requests/new) to report problems to us. Or you can post in the forum to communicate with other developers. Forum link: https://djisdksupport.zendesk.com/hc/zh-cn/community/topics
<br><p><em><small>°°°</small></em></p>

fire717, 2021-11-02:
And I have catkin_make Onboard-SDK-ROS, 
if I ignore the UserConfig.txt,
when I run 

> roslaunch dji_osdk_ros dji_sdk_node.launch

I will get this error

```setting /run_id to 307ad436-3bbc-11ec-91b0-18c04d2cbd6b
process[rosout-1]: started with pid [22277]
started core service [/rosout]
process[dji_sdk-2]: started with pid [22280]
[ INFO] [1635844000.185248872]: Advanced Sensing is Enabled on M210.
terminate called after throwing an instance of 'std::runtime_error'
  what():  User configuration file not found
[dji_sdk-2] process has died [pid 22280, exit code -6, cmd /home/yw/workshop/pipe/catkin_ws/devel/lib/dji_osdk_ros/dji_sdk_node __name:=dji_sdk __log:=/home/yw/.ros/log/307ad436-3bbc-11ec-91b0-18c04d2cbd6b/dji_sdk-2.log].
log file: /home/yw/.ros/log/307ad436-3bbc-11ec-91b0-18c04d2cbd6b/dji_sdk-2*.log
`

fire717, 2021-11-02:
可以了，现在的问题是

roslaunch dji_osdk_ros dji_sdk_node.launch

```auto-starting new master
process[master]: started with pid [22530]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 0213f356-3bbd-11ec-91b0-18c04d2cbd6b
process[rosout-1]: started with pid [22541]
started core service [/rosout]
process[dji_sdk-2]: started with pid [22544]
[ INFO] [1635844351.753671889]: Advanced Sensing is Enabled on M210.
Read App ID
User Configuration read successfully. 

[3756779.274]ERRORLOG/1 @ initVehicle, L215: Failed to initialize Linker channel
[dji_sdk-2] process has died [pid 22544, exit code -11, cmd /home/yw/workshop/pipe/catkin_ws/devel/lib/dji_osdk_ros/dji_sdk_node __name:=dji_sdk __log:=/home/yw/.ros/log/0213f356-3bbd-11ec-91b0-18c04d2cbd6b/dji_sdk-2.log].
log file: /home/yw/.ros/log/0213f356-3bbd-11ec-91b0-18c04d2cbd6b/dji_sdk-2*.log`

dji-dev, 2021-11-03:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/55594'>Zendesk ticket #55594</a>:<br><br>报错显示串口初始化失败，可以检查一下使用的串口设备和配置的串口设备名是否一致。
 [image]
关于串口相关，可以参考论坛相关帖：（OSDK-ROS是基于OSDK Linux的应用，原理上是一样的，可以参考来理解功能）
​https://sdk-forum.dji.net/hc/zh-cn/articles/900006796503-%E6%B3%9B%E8%B0%88DJI-OSDK%E4%B8%8E%E6%97%A0%E4%BA%BA%E6%9C%BA%E7%9A%84%E4%B8%B2%E5%8F%A3%E9%80%9A%E4%BF%A1<br><a href='https://djisdksupport.zendesk.com/attachments/token/nhgDUMbzyrnmqApUknHD4bK3U' target='zd_attach'>inline1469491901.png</a>
<br><p><em><small>°°°</small></em></p>