dji-sdk/Onboard-SDK-ROS #486: Drone version not obtained! Third party computer

Caseiber, 2021-11-08:
I am trying to use an Nvidia Jetson Nano as an onboard computer to communicate with the Matrice 300 RTK using the OSDK. The computer and drone are connected using the OSDK Expansion module. I am using the dual USB cable (USB 3 Type A) to connect to the USB port on the computer.

I am getting the error 
```
ERRORLOG/1 @ getDroneVersion, L1707: Drone version not obtained! Please do not proceed.
Possible reasons:
	Serial port connection:
		* SDK is not enabled, please check DJI Assistant2 -> SDK -> [v] Enable API Control.
		* Baudrate is not correct, please double-check from DJI Assistant2 -> SDK -> baudrate.
		* TX and RX pins are invert
[4290298.543]STATUS/1 @ functionalSetUp, L282: Shake hand with drone Fail ! Cannot get drone version. (1/20)
```
I have tried using all of the baud rates available in DJI Assistant 2 (changing them in both the app and the UserConfig.txt) and had no luck.

These are the current contents of my UserConfig.txt (without my actual app id and app key)

```
  app_id : [id]
  app_key : [key]
  device : /dev/ttyTHS1
  baudrate : 921600
  acm_port : /dev/ttyACM0
```
Thank you.

dji-dev, 2021-11-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/55973'>Zendesk ticket #55973</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The error message indicates that the serial port communication is abnormal. This problem is usually related to the serial port hardware or settings. It is recommended to refer to the following：
https://sdk-forum.dji.net/hc/en-us/articles/360062324793-OSDK-serial-communication-failure-troubleshooting-suggestions

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

Caseiber, 2021-12-03:
The solution to this problem was that I needed a serial connection and not just the usb connection. I did not realize this at first. I connected the TX, RX, and GND pins OSDK Expansion module to the corresponding pins on third party computer and that fixed the issue.

closed 2021-12-03 by Caseiber