dji-sdk/Onboard-SDK-ROS #488: Cannot get MOP data

hieudv-terradrone, 2021-11-10:
I'm implementing mop func for connecting and transfer data bw OSDK & MSDK.
I could connect its, but cannot get data from ```recvData``` func
Could any one give me some indication to do it? Thank you
```
bool VehicleNode::startSubscribeForGetDataMopCallback(SendDataMop::Request& request, SendDataMop::Response& response)
{
  ROS_INFO_STREAM("Start subscribe for geting MOP Data callback");
  if(ptr_wrapper_ == nullptr)
  {
    ROS_ERROR_STREAM("Vehicle modules is nullptr");
    return false;
  }
  MopPipeline *OM_Pipeline = NULL;
  bool ret = ptr_wrapper_->getMopPipeline(request.id, OM_Pipeline);
  if (!ret || OM_Pipeline == NULL) {
    response.result = ret;
    ROS_ERROR_STREAM("OM_Pipeline is nullptr");
    return 0;
  }
  pthread_t thread_publish;
  int result = pthread_create(&thread_publish, NULL, threadPublishMopDataTo, OM_Pipeline);
  if(result != 0) {
    DERROR("Server send mop data create failed!\n");
  } else {
    DSTATUS("Server send mop data create success!\n");
  }
  return response.result != 0;
}

void* VehicleNode::threadPublishMopDataTo(void *arg)
{
  MopPipeline *OM_Pipeline = (MopPipeline *)arg;
  if (!OM_Pipeline)
    std::runtime_error("Error param");
  uint8_t *recvBuf;
  const uint8_t RELIABLE_RECV_ONCE_BUFFER_SIZE = 100 * 1024;
  recvBuf = (uint8_t *)OsdkOsal_Malloc(RELIABLE_RECV_ONCE_BUFFER_SIZE);
  if (recvBuf == NULL) {
    DERROR("[File-Service]  OsdkOsal_Malloc recv buffer error");
    return NULL;
  }
  MopPipeline::DataPackType readPack = {(uint8_t *) recvBuf, RELIABLE_RECV_ONCE_BUFFER_SIZE};
  // OsdkOsal_TaskSleepMs(5000);
  // receive data
  MopErrCode mopRet = MOP_PASSED;
  while(ros::ok())
  {
    memset(recvBuf, 0, RELIABLE_RECV_ONCE_BUFFER_SIZE);
    readPack.length = RELIABLE_RECV_ONCE_BUFFER_SIZE;
    mopRet = OM_Pipeline->recvData(readPack, &readPack.length);

    if (mopRet != MOP_PASSED) {
      if (mopRet == MOP_TIMEOUT) {
      DSTATUS("[File-Service] receive timeout");
      } else if (mopRet == MOP_CONNECTIONCLOSE) {
          DSTATUS("[File-Service]  connection of pipeline [%d] closed", OM_Pipeline->getId());
          DSTATUS("[File-Service]  disconnect from cilent now stop task");
          break;
      } else {
          DERROR("[File-Service]  recv data from client error,stat:%lld", mopRet);
      }
      OsdkOsal_TaskSleepMs(1000);
    } else {
        if (readPack.length > 0) {
          DSTATUS("Receive buf {%d ...}", recvBuf[0]);
          // publish 
          dji_osdk_ros::MopData mop_data;
          mop_data.data.resize(readPack.length);
          for (int i=0; i<readPack.length; i++)
          {
            mop_data.data[i] = readPack.data[i];
          }
          mop_publisher_.publish(mop_data);
        }
    }
  }
}
```

dji-dev, 2021-11-10:
Agent comment from DJI SDK in <a href='https://djisdksupport.zendesk.com/agent/tickets/56100'>Zendesk ticket #56100</a>:<br><br>尊敬的开发者，感谢您联系DJI 大疆创新
由于github不是我们主要的咨询渠道，您的问题可能跟进不及时。我们建议您通过填写表单（ https://djisdksupport.zendesk.com/hc/zh-cn/requests/new ）向我们反馈问题。或者您也可以在论坛发帖，与其它开发者交流。论坛链接：https://djisdksupport.zendesk.com/hc/zh-cn/community/topics

Dear developer, thank you for contacting DJI.
Since github is not our main consultation channel, your questions may not be followed up in time. We recommend that you fill in the form (https://djisdksupport.zendesk.com/hc/en-us/requests/new) to report problems to us. Or you can post in the forum to communicate with other developers. Forum link: https://djisdksupport.zendesk.com/hc/zh-cn/community/topics
<br><p><em><small>°°°</small></em></p>

hieudv-terradrone, 2021-11-12:
I find the issue, my stupid code
`const uint8_t RELIABLE_RECV_ONCE_BUFFER_SIZE = 100 * 1024;`
I could send/recv msg from MSDK with ROS. Thank you so much for support!

closed 2021-11-12 by hieudv-terradrone