dji-sdk/Onboard-SDK-ROS #493: Waypoint Mission using RTK

dregner, 2022-01-11:
Hello there,

I am developing an autonomous mission using Waypoint Mission at M210 V2 RTK using Manifold 2G embedded with Onboard-SDK-ROS 3.8.1 and Onboard-SDK 3.9. I would like to know if there is an option to set Waypoint Mission use RTK's data when it is available. If so, where can I set the data provided from GPS to RTK to control the mission?

Thanks,
Best regards

dji-dev, 2022-01-12:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/59093'>Zendesk ticket #59093</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The OSDK application cannot switch GPS to RTK directly . If RTK is enabled and fixed on the remote control APP, the drone will use RTK to locate during  waypoint mission.

And there are some TOPICs about RTK used to get the RTK data from the drone.
DJI::OSDK::Telemetry::TOPIC_RTK_POSITION,
DJI::OSDK::Telemetry::TOPIC_RTK_VELOCITY,
DJI::OSDK::Telemetry::TOPIC_RTK_YAW,
DJI::OSDK::Telemetry::TOPIC_RTK_POSITION_INFO,
DJI::OSDK::Telemetry::TOPIC_RTK_YAW_INFO,

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

dregner, 2022-01-12:
Thank you for quickly responde. Unfortunatelly when I tried to use waypoint mission when RTK is enable and fixed the the drone still was using GPS data, I was creating waypoints based on RTK data but as the altitude from GPS and RTK was different and drone was reaching it based on GPS data.

Unfortunatelly If there is no option to manual switch the data provided from different GNSS sensors, I will close this issue.

closed 2022-01-12 by dregner