dji-sdk/Onboard-SDK-ROS #496: Waypoint Mission unrecorded home point

wongkenchiang97, 2022-02-20:
I'm testing DJI demo_mission with below setups:
1. ROS OSDK version: 3.8.1
2. OS: Ubuntu 20.04
3. Flight Controller(DJI M100): N1 

My waypoints are start location->selected goal location->start location. 3 waypoints.
[image]

Terminal output of demo_mission node. Warning message:"Failed sending mission start command".
[image]

Terminal output of DJI ROS SDK node. ROS OSDK status:"MISSION_UNRECORDED_HOME".
[image]

I've tried to add home point on DJI ROS SDK node's code referring to #526 issue, but SDK package don't have 
"<dji_flight_controller.hpp>" header file.

Clarification of positioning and health module:
I've change my ROS SDK Node's positioning and health module. Instead of publishing position and health information from the serial bus to ROS platform, It will publish position and health value from other positioning module to ROS platform. Therefore, the GPS signal of the drone may not be strong on this testing.

Anyone have solved this issue? @rohitsantdji @QRI-nashw @rcabg @trgiman @rocklinsuv

dji-dev, 2022-02-21:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/60472'>Zendesk ticket #60472</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The info you provide shows that the latitude and longitude of your waypoint is an angle value. please upload with radians.
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

wongkenchiang97, 2022-02-21:
> 

Thanks for your kind reply and reminder!!