dji-sdk/Onboard-SDK-ROS #498: m600 pro 妙算2g osdk3.8.1 报错

jiangqianpeng, 2022-03-05:
运行roslaunch dji_sdk sdk.launch时，报错如下图，是什么问题呢

dji-dev, 2022-03-05:
Agent comment from DJI SDK in <a href='https://djisdksupport.zendesk.com/agent/tickets/61095'>Zendesk ticket #61095</a>:<br><br>尊敬的开发者，感谢您联系DJI 大疆创新
由于github不是我们主要的咨询渠道，您的问题可能跟进不及时。我们建议您通过填写表单（ https://djisdksupport.zendesk.com/hc/zh-cn/requests/new ）向我们反馈问题。或者您也可以在论坛发帖，与其它开发者交流。论坛链接：https://djisdksupport.zendesk.com/hc/zh-cn/community/topics

Dear developer, thank you for contacting DJI.
Since github is not our main consultation channel, your questions may not be followed up in time. We recommend that you fill in the form (https://djisdksupport.zendesk.com/hc/en-us/requests/new) to report problems to us. Or you can post in the forum to communicate with other developers. Forum link: https://djisdksupport.zendesk.com/hc/zh-cn/community/topics
<br><p><em><small>°°°</small></em></p>

jiangqianpeng, 2022-03-05:
怎么插入图片呢？

jiangqianpeng, 2022-03-05:
怎么插入图片呢？