dji-sdk/Onboard-SDK-ROS #505: Services are bad for low level flight control, Topics needed

xaedes, 2022-06-22:
Previously it was possible to set low level flight controller target via topics.
Now this seems to be entirely replaced by services.

This may make sense for high level control. 
But an RPC for low level control, like drone velocity, is just too much.
The response result is irrelevant and you want high throughput with minimal latency and overhead.

Services only add overhead here and also make it *impossible to record* sent flight commands with rosbag.

It really is **big** step backwards.

snakehaihai, 2022-08-11:
maybe that's what they want. I haven`t switched to 4+ yet. Still using 3.9 for this very reason :) . I only tried 4 with high level waypoint control.  If you a researcher, then stay at 3.9.

xaedes, 2022-09-08:
I now just added Flight Control via topic. 

Last year we had the same problem for gimbal control, where we wanted it to smoothly follow a tracked target. With the service based control it was super choppy and barely useable at all. Right after switching to topic based control it was as smooth as a polished marble ball. 

https://github.com/xaedes/Onboard-SDK-ROS/commit/a269123ca6a31ceef7aae542e18392f4a8ab01a6

snakehaihai, 2022-09-14:
well belive it or not. most people are moving back to pixhawk for this very reason.  https://www.youtube.com/watch?v=MxeoKVXrmEs&ab_channel=MARSLABHKU

[image]

Threepain, 2023-01-04:
I really agree with your suggestion. Some low-level flight control commands can be released through topic. Maybe Payload-SDK is what DJI really wants us to use, the last update of Onboard-SDK-ROS was two years ago (:D).

snakehaihai, 2023-01-08:
> I really agree with your suggestion. Some low-level flight control commands can be released through topic. Maybe Payload-SDK is what DJI really wants us to use, the last update of Onboard-SDK-ROS was two years ago (:D).

Don't get to the psdk 3.3 series ... that shit is missing tons of cmakelist and no ros interface. Basically, there is only one demo which is connecting to the drone. Need video feed? ok put in cmake urself and make hardware connector yourself.  Need quite a lot of development to make it workable.