dji-sdk/Onboard-SDK-ROS #507: M210 V2 main camera and FPV camera service failure

dregner, 2022-07-13:
After initiate dji_sdk node to m210 aircraft with advanced sensing I am calling rosservice `/dji_sdk/setup_camera_stream` to initialize X5S and FPV camera stream but its result in same error. When activate main camera, it starts to stream at RC but not at `/dji_sdk/main_camera_images` topic and when I acquire a image or video it crash. Same error occur when I asked to stream FPV camera, it crashes before stream.

### Output errors
- X5S error when taking photo:
```
STATUS/1 @ keep_camera_x5s_state, L44:
STATUS/1
STATUS/1
STATUS/1
STATUS/1 Assertion desc failed at src/libswscale/swscale_internal.h:668
[dji_sdk-2] process has died [pid 11226, exit code -6, cmd /home/jetson/catkin_ws/devel/lib/dji_sdk/dji_sdk_node __name:=dji_sdk __log:=/home/jetson/.ros/log/f3571ae0-02aa-11ed-a4cc-00044be607a6/dji_sdk-2.log].
log file: /home/jetson/.ros/log/f3571ae0-02aa-11ed-a4cc-00044be607a6/dji_sdk-2*.log
```

- FPV image when requesting stream:
 ```
[h264 @ 0x7f58018330] no frame!
[h264 @ 0x7f58018330] non-existing PPS 63 referenced
[h264 @ 0x7f58018330] non-existing PPS 63 referenced
[h264 @ 0x7f58018330] decode_slice_header error
[h264 @ 0x7f58018330] non-existing PPS 63 referenced
[h264 @ 0x7f58018330] decode_slice_header error
[h264 @ 0x7f58018330] non-existing PPS 63 referenced
[h264 @ 0x7f58018330] decode_slice_header error
[h264 @ 0x7f58018330] non-existing PPS 63 referenced
[h264 @ 0x7f58018330] decode_slice_header error
[h264 @ 0x7f58018330] no frame!
Assertion desc failed at src/libswscale/swscale_internal.h:668
[dji_sdk-2] process has died [pid 11711, exit code -6, cmd /home/jetson/catkin_ws/devel/lib/dji_sdk/dji_sdk_node __name:=dji_sdk __log:=/home/jetson/.ros/log/145a7d54-02ab-11ed-9fdd-00044be607a6/dji_sdk-2.log].
log file: /home/jetson/.ros/log/145a7d54-02ab-11ed-9fdd-00044be607a6/dji_sdk-2*.log
```

It looks problem of FFmpeg version `Assertion desc failed at src/libswscale/swscale_internal.h:668`

### Onboard PC configuration:

- Jetpack 4.6 - Ubuntu 18.04
- CUDA 10.2
- OpenCV 3.3.1 parallel installed with 4.5.5
- OSDK 3.9
- ROS Melodic
- OSDK ROS 3.8.1
- FFmpeg version 2.8

dji-dev, 2022-07-14:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/67596'>Zendesk ticket #67596</a>:<br><br>The error message is ffmpeg decoding assertion error. For M210 v2 model, it is recommended to use OSDK 4.1 + OSDK ROS 4.1. And he sample is based on ffmpeg version 2.8.15. Please confirm the ffmpeg version again.
<br><p><em><small>°°°</small></em></p>

dregner, 2022-08-24:
Ok, unfortunatelly I could not solve with OSDK 3.9 and migrated to 4.1 and all worked just fine.

closed 2022-08-24 by dregner