dji-sdk/Onboard-SDK-ROS #509: cannot find dji_vehicule.hpp file

abdouakhad, 2022-08-16:
Hi Everyone.
I have already followed these steps
`mkdir build`
`cd build`
`cmake ..`

But Once I run the `make` command, I have the error below.

`Building CXX object src/dji_osdk_ros/modules/CMakeFiles/dji_osdk_ros.dir/dji_vehicle_node_mission_services.cpp.o
In file included from /home/caytu/Onboard-SDK-ROS/src/dji_osdk_ros/modules/dji_vehicle_node_mission_services.cpp:29:
/home/caytu/Onboard-SDK-ROS/include/dji_osdk_ros/dji_vehicle_node.h:34:10: fatal error: dji_vehicle.hpp: No such file or directory
   34 | #include <dji_vehicle.hpp>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/dji_osdk_ros/modules/CMakeFiles/dji_osdk_ros.dir/build.make:63: src/dji_osdk_ros/modules/CMakeFiles/dji_osdk_ros.dir/dji_vehicle_node_mission_services.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:5710: src/dji_osdk_ros/modules/CMakeFiles/dji_osdk_ros.dir/all] Error 2
make: *** [Makefile:141: all] Error 2`.

Does anyone know why?

dji-dev, 2022-08-17:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/69325'>Zendesk ticket #69325</a>:<br><br>Have you executed "make install"
https://developer.dji.com/document/e34c3fc4-dc0e-4cbf-b7c0-589b6e941a01
 [image]<br><a href='https://djisdksupport.zendesk.com/attachments/token/SkFRg0sVNipnXww1lwvHwcLSq' target='zd_attach'>image.png</a>
<br><p><em><small>°°°</small></em></p>

abdouakhad, 2022-08-17:
Yes I ran exactly `sudo make -j7 install`

closed 2022-09-29 by abdouakhad