dji-sdk/Onboard-SDK-ROS #511: help DJi Matrice 100 Ros with Raspberry Pi walthrough guide

Matthewsw1234, 2022-09-21:
Hi everyone. I know the Matrice 100 is a dated platform by now but I was wanting to develope some software for it. Could anyone give me a new guide to connecting the drone to a raspberry pi and do a walkthrough of the whole setup. Right now I just have a pi with no software on it besides the Rasp Os. Honestly i'm new to prgramming and know next to nothing about what I am doing. If someone could either guide me from step one to the end on how to do this it would be great. Once everything is connected and working with simple commands like "takeoff" "landing" etc it would be great. I would be ok but all current guides for this are either dated and don't work or are running different software.

Thanks!

dji-dev, 2022-09-22:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/70966'>Zendesk ticket #70966</a>:<br><br>M100 is recommended to use OSDK 3.9 version. If you need to use ROS, OSDK linux 3.9 + OSDK ROS 3.8 is recommended. The offline documentation can be downloaded from the developer website. There are tutorials on hardware connection and SDK sample compilation.
https://developer.dji.com/document/1be2b992-5862-4eb5-819b-dd6036f76133
 [image]
<br><p><em><small>°°°</small></em></p>

Matthewsw1234, 2022-09-23:
> Agent comment from kyle.cai in [Zendesk ticket #70966](https://djisdksupport.zendesk.com/agent/tickets/70966):M100 is recommended to use OSDK 3.9 version. If you need to use ROS, OSDK linux 3.9 + OSDK ROS 3.8 is recommended. The offline documentation can be downloaded from the developer website. There are tutorials on hardware connection and SDK sample compilation. https://developer.dji.com/document/1be2b992-5862-4eb5-819b-dd6036f76133 [image]
> 
> _°°°_

Hi. Thanks for responding. I see that but it is hard to follow as most of the links and pictures don’t work anymore. What I’m mainly trying to figure out is if I’m using a Raspberry Pi3 B, what OS and software do I use?

closed 2022-11-06 by Matthewsw1234