dji-sdk/Onboard-SDK-ROS #513: ROS launch ttyS0 not working

Matthewsw1234, 2022-10-12:
I keep trying to get the drone to connect to the Raspberry pi and can’t get it to work. I’m using Ubuntu 20.04 LTS on a Pi 3 B. This is the error I get. Sorry for the photo. 
[image]

Matthewsw1234, 2022-10-13:
[image]

dji-dev, 2022-10-13:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/71848'>Zendesk ticket #71848</a>:<br><br>Please confirm that the ttyS0 is available on your onboard computer and that the ttyS0 is connected to the drone TTL. If the device name is correct, please check the serial port access permission.
 [image]<br><a href='https://djisdksupport.zendesk.com/attachments/token/2PegwxjboIsj4wD35qTIcAqeQ' target='zd_attach'>image.png</a>
<br><p><em><small>°°°</small></em></p>

Matthewsw1234, 2022-10-15:
> Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/71848'>Zendesk ticket #71848</a>:<br><br>Please confirm that the ttyS0 is available on your onboard computer [...]

Hi. Thanks for responding. I have checked and the TTYS0 is reading stuff from the drone and I am a part of the Dialout group. Am I missing something?

Matthewsw1234, 2022-10-18:
So I updated to 3.8 and still have the same problem. Not sure why

Matthewsw1234, 2022-11-06:
For anyone who might need this, the getty.service for my TTYS0 port was a n the way. Just stop it and disable it, then use sudo chmod 777 /dev/ttyS0