dji-sdk/Onboard-SDK-ROS #516: waypoint ? google map ?

HunnyRasheed, 2022-10-24:
is there any easy way to create waypoints for ros osdk? as if we compare it to msdk it was a lot easy to generate waypoints in the mobile application but how to generate it for ros osdk ? is there any way to create google api which can be linked with our code ?

dji-dev, 2022-10-25:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/72378'>Zendesk ticket #72378</a>:<br><br>For the M300, the OSDK side uploads the waypoint coordinates through the API. Developers need to develop and generate the waypoints by themselves. For example, import the waypoint file, parse the coordinates in the file and then upload it through the OSDK API.
<br><p><em><small>°°°</small></em></p>

HunnyRasheed, 2022-10-25:
can I use tktiner to make waypoints ?

HunnyRasheed, 2022-10-25:
[image]

is there any possible way to make this type of api in ros ? or any program which we can implement this ?

peaky-quest, 2022-12-03:
> This is not a question related to this ROS package but I will give you some insight on how to move forward.
>
> You can use folium or requests on a map API (such as Google maps or Mapbox) to get a map image in Python and present it in a GUI. Then, using some maths transform pixel coordinates to latitude and longtitude. I think it is easier to integrate ROS with Qt instead of Tkinter in Python. Then you need a way to pass the generated coordinates from Qt to a waypoint mission and send it to the aircraft. A solution would be to save these coordinates in a file and then read them from a custom version of waypointV2_node or just transmit them using a ROS message.

Thankyou for the suggestion, I will look into it.