dji-sdk/Onboard-SDK-ROS #518: Drone speed not changing in each waypoint

frank20a, 2022-11-10:
I am running a way point mission where the global auto and max flight are set as follows:
```cpp
initWaypointV2Setting_.request.waypointV2InitSettings.maxFlightSpeed = 10;
initWaypointV2Setting_.request.waypointV2InitSettings.autoFlightSpeed = 7;
```
and for each waypoint is set indipendently and dynamically from file input:
```cpp
setWaypointV2Defaults(waypointV2);
waypoint_file >> waypointV2.latitude >> waypointV2.longitude >> waypointV2.relativeHeight >> waypointV2.heading >> waypointV2.maxFlightSpeed >> waypointV2.autoFlightSpeed;
waypointList.push_back(waypointV2);
```
I later print the `waypointList` and get the following output confirming that the waypoints have been correctly created:
```cpp
int i = 0;
for(auto wp : waypointList){
  ROS_INFO("Waypoint %d: %f, %f, %f, %f, %f, %f", i++, waypointV2.latitude, waypointV2.longitude, waypointV2.relativeHeight, waypointV2.heading, waypointV2.maxFlightSpeed, waypointV2.autoFlightSpeed);
}
Waypoint 0: XX.XXX, YY.YYY, 10.0, 0.0, 9.0, 5.0
Waypoint 1: XX.XXX, YY.YYY, 35.0, 0.0, 7.0, 3.0
Waypoint 2: XX.XXX, YY.YYY, 35.0, 0.0, 7.0, 5.0
```
This is `setWaypointV2Defaults` (as in waypointV2_mission.cpp):
```cpp
void setWaypointV2Defaults(dji_osdk_ros::WaypointV2 &waypointV2)
{
  // Flight mode (direct point 2 point)
  waypointV2.waypointType = dji_osdk_ros::DJIWaypointV2FlightPathModeGoToPointInAStraightLineAndStop;
  // Heading Mode (will face direction of next point)
  waypointV2.headingMode = dji_osdk_ros::DJIWaypointV2HeadingModeAuto;
  waypointV2.config.useLocalCruiseVel = 1;
  waypointV2.config.useLocalMaxVel = 0;

  waypointV2.dampingDistance = 40;
  waypointV2.heading = 0; // Final orientation
  waypointV2.turnMode = dji_osdk_ros::DJIWaypointV2TurnModeClockwise;

  // (NOT SURE) if heading is DJIWaypointV2HeadingTowardPointOfInterest the drone will always face this point:
  waypointV2.positionX = 0;
  waypointV2.positionY = 0;
  waypointV2.positionZ = 0;
  waypointV2.maxFlightSpeed = 9;
  waypointV2.autoFlightSpeed = 2;
}
```

I have played arround with:
```cpp
  waypointV2.config.useLocalCruiseVel = 1;
  waypointV2.config.useLocalMaxVel = 0;
```
and using `useLocalCruiseVel=0` and `useLocalMaxVel=0` the drone only follows the global auto speed and travels to every waypoint with 7m/s. Using any combination of `useLocalCruiseVel=1` and/or `useLocalMaxVel=1` the drone only follows the first waypoint auto speed and travels to every waypoint with 5m/s even though Waypoint1 has an autoFlightSpeed of 3m/s.

Is there a way for the drone to follow different speeds per waypoint? For example here it should move to Waypoint0 with 5m/s then to Waypoint1 with 3m/s and then to Waypoint2 with 5m/s again.

dji-dev, 2022-11-11:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/73275'>Zendesk ticket #73275</a>:<br><br>Only the global cruise speed can be set in the OSDK waypoint mission, and it is not supported to set different speeds for a single waypoint. You can consider resetting the global cruise speed when you reach the waypoint.
<br><p><em><small>°°°</small></em></p>

frank20a, 2022-11-29:
> 

Would I have to stop and reupload the rest of the mission each time I want to change the cruise speed? Also is there a way to change how much time the aircraft waits on each waypoint?

dji-dev, 2022-11-30:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/73275'>Zendesk ticket #73275</a>:<br><br>There is no need to stop and re-upload, and the cruise speed can be dynamically changed during the mission.
Hover time can be set by adding an action.
How to set waypoint action hover in waypoint V2?(https://sdk-forum.dji.net/hc/en-us/articles/900005820706)
<br><p><em><small>°°°</small></em></p>

closed 2022-12-14 by frank20a