dji-sdk/Onboard-SDK-ROS #519: Gimbal Control does not react before every resetting

engcang, 2022-11-16:
Hi. 
I am using M300-RTK drones and Gremsy PIXY SM gimbal with it.

With DJI OSDK-ROS, I can control gimbal angle with rosservice, but sadly, I only can adjust/control the gimbal angle with /gimbal_task_control service and then it never rotates. More weirdly, it rotates again after every resetting with "is_reset: true" within same /gimbal_task_control service.

I want to continuously adjust the angle of gimbal but currently I have to reset before rotate to the desired angle.

I will look forward to the your great solution.

Thanks.

dji-dev, 2022-11-17:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/73590'>Zendesk ticket #73590</a>:<br><br>Is Gremsy PIXY SM gimbal developed through DJI PSDK? Is it possible to control the rotation directly using the remote control?
<br><p><em><small>°°°</small></em></p>

engcang, 2022-11-17:
@dji-dev 
Hi.

Yes I can directly control gimbal using remote controller and I can see it is recognized as PIXY SM on the screen of the controller.

dji-dev, 2022-11-17:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/73590'>Zendesk ticket #73590</a>:<br><br>Please provide the rotation parameter settings.
<br><p><em><small>°°°</small></em></p>

closed 2022-11-22 by engcang