dji-sdk/Onboard-SDK-ROS #520: How to Set Gimbal Reference Point for H20T?

jin-bae-joulea, 2023-02-20:
I have an H20T hooked up to my M300 and I'm using OSDK-ROS on my onboard Jetson to control the gimbal. However, I've noticed that the angle in the default position (facing directly front) is 0 degree roll 0 degree pitch and -100 degree yaw. Is there a way to set this angle to be 0 0 0 at its default position?

dji-dev, 2023-02-21:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/77732'>Zendesk ticket #77732</a>:<br><br>The gimbal attitude angle obtained by OSDK is based on NEU, and 0 degrees refers to north.
<br><p><em><small>°°°</small></em></p>

jin-bae-joulea, 2023-02-23:
> Caution: This is an external email and may be malicious. Please take care when clicking links or opening attachments.
>
> Agent comment from kyle.cai in Zendesk ticket #77732<https://djisdksupport.zendesk.com/agent/tickets/77732>:

Does the magnetometer reading only affect the initial gimbal yaw value or is the gimbal yaw value entirely based on the magnetometer reading according to NEU? We're talking about dji_osdk_ros/gimbal_angle topic right?