dji-sdk/Onboard-SDK-ROS #522: M100_initialization_issue

Aztec02, 2023-04-02:
I am trying to run a ros demo with M100 + Raspberry Pi Model B on simulation.
I have the M100 turned on and connected with a USB-to-Serial to Raspberry.
The remote controller turned on and connected with the cell phone with internet connection and DJI GO open.
I run the Simulation on DJI Assistant 2 (SDK API enabled), but when I launch the dji_sdk node  I get the following error.
[image]

Any ideas what might be wrong?

Aztec02, 2023-04-09:
I also attach the ros log.

(BTW the osdk sample is running OK)

[c1fe08cc-d703-11ed-a323-ed5940d2bbcb.zip](https://github.com/dji-sdk/Onboard-SDK-ROS/files/11186091/c1fe08cc-d703-11ed-a323-ed5940d2bbcb.zip)

frankleeuse, 2023-04-16:
I'm using phantom3 and I can't find ttyUSB0, how did you find your ttyUSB0